CR6Community/Marlin #1: Clean-up touch screen code

Sebazzz, 2020-09-25:
The touch screen code is a mess currently:

- Switching between languages is sprinkled throughout the source
- There is one big switch to handle events from the touch UI

Sebazzz, 2020-09-26:
Support for Chinese is dropped. The giant switch for event handling still needs love.

Sebazzz, 2020-09-26:
@JonBradbury @njoythegame I've been looking though the Marlin sources and there is already stuff in place to communicate with DGUS displays:

https://github.com/Sebazzz/Marlin/tree/creality-cr6-merge-attempt/Marlin/src/lcd/extui/lib/dgus
https://github.com/Sebazzz/Marlin/blob/creality-cr6-merge-attempt/Marlin/src/lcd/extui/dgus_lcd.cpp

I think this can be basis for re-implementing the Creality LCD interfacing code. It is essentially many event handlers for all things that happen in Marlin.

Sebazzz, 2020-09-27:
I've been looking into using this as a basis, but the ExtUI implementation for DGUS is quite opinionated:

- In the SD file list you should scroll
- Only one file list instead of five fixed pages
- No separate print status screen for paused / running print

We can adapt the touch screen to use this, but I think for now this is a bridge too far as that would need non-trivial changes to the touch screen. I would rather not do that at the moment because we're also still trying to get basic functions working, the touch screen configuration is all binary and unmergable, so it is best not to deviate too much from mainline.

Sebazzz, 2020-09-27:
@JonBradbury: I did a write-up here: https://discord.com/channels/759603746270609428/759605549305692170/759870235838251008

I think this might be a viable way to continue.

chriskoups, 2020-10-01:
Can you share the discord link?

Sebazzz, 2020-11-01:
No longer happens in extui. I have been bed leveling extensively.

Ready for next release.

Sebazzz, 2020-11-01:
@chriskoups Sorry - this: https://discord.gg/zzrBay73

commit f8af7125 (2020-09-25): Drop support for Chinese. Fixes #1

The touch screen still has the concept of English and Chinese, but
we only change to English pages, and switch only to English icons.

The settings are still there visually, but it will be taken care of in Sebazzz/CR-6-touchscreen#2.
closed 2020-11-01 by Sebazzz