CR6Community/Marlin #4: Extruder feed rate is unusable slow through LCD touch screen

NickDevoctomy, 2020-09-26:
### Description

Extruder feeding through LCD touch screen is extremely slow

### Steps to Reproduce

Attempt to feed 60mm using menu, it takes a whole minute

1. Navigate to feed menu on touch screen
2. Enter 60mm
3. Press feed

Personally I would expect it to feed at a reasonable rate, preferable the same speed as Octoprint, which operates at 300 mm/m

It operates at 60 mm/m making the feature unusable for loading / unloading filament

#### Additional Information

I also noticed that in the source code there are some poorly named enums,

PROC_COM

HeaterMoveEnterKey      = 17,
HeaterMoveStartKey      = 18,

These have nothing to do with the heater, but rather the Axis, so should ideally be

AxisMoveEnterKey      = 17,
AxisMoveStartKey      = 18,

Line 63 in touch_lcd.cpp appears to hold the feed rates for the axis movement feature.  And the last value for the extruder is set to 60, my recommendation would be (5*60) to match Octoprint.

Obviously I understand that this could be entirely subjective so my apologies if you don't care about this.  I just thought I'd mention it whilst I'm poking about in the source.

Sebazzz, 2020-09-26:
Is this something you experience in stock firmware as well?

NickDevoctomy, 2020-09-26:
Yes, it was like that on the printer, out-of-the box if that's what you mean, along with many other issues / bugs.

Sebazzz, 2020-09-26:
Alright. I can't promise I can do anything anytime soon though, I have still yet to receive my unit and play with the touch screen how this particular part works/behaves 😄

NickDevoctomy, 2020-09-26:
Sure thing, no worries, as I mentioned above this could be seen as quite subjective as I'm not sure how many other people rely on the same features that I do. I've put it into my own fork of your repo so am giving it a good test.  Understood if you don't want to put it in.

Hopefully you receive it soon! Mine was sent on 1st of August and I got it last week (UK).  The tracking info was sitting in Germany for most of that.

64Spaces, 2020-10-06:
Yes, the filament feed rate, in or out, is painfully slow. Since the extruder is not visible, I have to feel the filament to be sure it’s moving.

Sebazzz, 2020-10-25:
> I also noticed that in the source code there are some poorly named enums,
> 
> PROC_COM
> 
> HeaterMoveEnterKey = 17,
> HeaterMoveStartKey = 18,

Heh... yeah this is the legacy Creality left us. This is because the virtual pointers are used across the screens. What might be an "enter" key on one screen is a button to abort the print on another. The real distinction between the codes is based on the button value. This also makes the code very hard to refactor.

Check this for more information: https://github.com/CR6Community/CR-6-touchscreen#how-buttons-are-handled-with-code

Sebazzz, 2020-11-01:
@nickacker I think you said this is the feed rate, didn't you?

Sebazzz, 2020-11-05:
Extui uses 6mm/s resulting in 360mm/m.

closed 2020-11-05 by Sebazzz

64Spaces, 2020-11-05:
Are we stuck at this feed rate when using the LCD?

Sebazzz, 2020-11-05:
Yes and no, you can change `FILAMENT_CHANGE_FAST_LOAD_FEEDRATE` and compile it yourself. Also, if you have a better suggestion for the feedrate, please do give it.

commit 9f6530a6 (2020-09-26): Fix preheat settings not put in UI on startup. Fixes #4