CR6Community/Marlin #5: Question: How are buttons added to the LCD touch screen?

NickDevoctomy, 2020-09-26:
Hi there, my apologies if this is not the correct place to ask this, in which case please disregard.

I was just poking about with your CR6-SE patches for Marlin, great work by the way, my printer is running pretty good now that you appear to have fixed a lot of the issues I had.

I wanted to add something to the feed menu, a filament load / unload option which I would previously have used via the M700 / M701 options through the old LCD screen, but I understand that this isn't possible with the new touch screen, even with  ADVANCED_PAUSE_FEATURE and FILAMENT_LOAD_UNLOAD_GCODES for one reason or another.

Anyway, is it possible to add buttons or does that require compilation of something that Creality hasn't provided?  I mean I have a zip with a bunch of bitmaps in and various bin files, but no idea how the bin files were generated.  Any ideas?

Thanks for your time.

Sebazzz, 2020-09-26:
We are trying to figure out how to extend the touch screen with exactly this project: https://github.com/Sebazzz/CR-6-touchscreen/

You need the DGUS software as stated in the README.md file there. It should be possible to add buttons and even new pages, but we have not tried that yet. Note that these are ugly binary unmergable files, so any work being done needs to be synchronized - and the implementation in `touch_lcd.cpp` is tightly coupled to the touch screen interface.

NickDevoctomy, 2020-09-26:
That's great, thanks so much for the info, I'll take a look now

closed 2020-09-26 by NickDevoctomy