CR6Community/Marlin #8: LCD Nozzle tempature command issues?

randomgitname, 2020-09-29:
### Description

Nozzle Tempature set via touchscreen triggers thermal runaway screen.
 
### Steps to Reproduce

1. Set nozzle temp to 220. Correct temp goal is set.
2. Heating to successfully to about 210, then stalls.
3. Thermal run away screen.

**Expected behavior:** [What you expect to happen]
Heat to 220.
**Actual behavior:** [What actually happens]
Heat to 210. Thermal runaway screen.
#### Additional Information
When printing from gcode or using the pla preheat LCD command heating seems to function as intended.

chriskoups, 2020-10-01:
Can you connect an octopi/computer or else, in order to see what command the printer receives ones you set the temperature from the touchscreen?
If you set the temperature higher, does it go again up to 210 and stalls?

randomgitname, 2020-10-04:
I Tried to dupicate this issue after using octoprint for a few days, and was unable to reproduce. perhaps the thermal runaway timer is a little tight or because I had my silicon sock off + different room temps or all of these things. closing.

closed 2020-10-04 by randomgitname