CR6Community/Marlin #10: Error 'Probing failed' when doing an second G29

Ranney1, 2020-10-04:
Printer is reporting an error when doing an second G29. (for example, for Better Visualiser)

1. Connect to printer
2. Home (G28)
3. Level (G29)
 > Everything Ok for now
4. Level (G29)
 > ERROR

If doing an G28 between the G29 then it is OK.

Error from printer:
Send: G29
Recv:  T:23.12 /0.00 B:31.02 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:23.12 /0.00 B:30.94 /0.00 @:0 B@:0
**Recv: Error:Probing Failed**
Changing monitoring state from "Operational" to "Error: Probing Failed"
Send: M112
Send: N2 M112*35
Send: N3 M104 T0 S0*34
Send: N4 M140 S0*97
Changing monitoring state from "Error: Probing Failed" to "Offline (Error: Probing Failed)"
Connection closed, closing down monitor
OCTO: Power off printer outlet

Ranney1, 2020-10-07:
Compiled debug firmware. This is the debugging info for the second probing. I see this problem reported on facebook multiple times. Anybody who has an idea what can be the problem?

Send: G29
Recv: >>> G29  X117.00 Y117.00 Z3.07
Recv: Machine Type: Cartesian
Recv: Probe: FIX_MOUNTED_PROBE
Recv: Probe Offset X0.00 Y0.00 Z0.20 (Aligned With & Above Nozzle)
Recv: Auto Bed Leveling: BILINEAR (enabled)
Recv: Z Fade: 2.00
Recv: ABL Adjustment Z+0.221 ( 0.000)
Recv: >>> do_blocking_move_to  X117.00 Y117.00 Z3.07
Recv: >  X117.00 Y117.00 Z10.00
Recv:  T:21.37 /0.00 B:21.64 /0.00 @:0 B@:0
Recv: <<< do_blocking_move_to  X117.00 Y117.00 Z10.00
Recv:   current_position= X117.00 Y117.00 Z10.00 : Leveling ON
Recv:   current_position= X117.00 Y117.00 Z10.00 : ...Now OFF
Recv:   current_position= X117.00 Y117.00 Z10.00 : sync_plan_position
Recv:   current_position= X117.00 Y117.00 Z10.00 : Probe::set_deployed
Recv: deploy: 1
Recv: Probe::do_z_raise(10.00)
Recv: >>> do_blocking_move_to  X117.00 Y117.00 Z10.00
Recv: >  X117.00 Y117.00 Z10.00
Recv: <<< do_blocking_move_to  X117.00 Y117.00 Z10.00
Recv: >>> do_blocking_move_to  X117.00 Y117.00 Z10.00
Recv: >  X117.00 Y117.00 Z10.00
Recv: <<< do_blocking_move_to  X117.00 Y117.00 Z10.00
Recv: >>> Probe::probe_at_point  X117.00 Y117.00 Z10.00
Recv: ...(30.00, 30.00, raise, 0, probe_relative)
Recv:   current_position= X117.00 Y117.00 Z10.00 :
Recv: >>> do_blocking_move_to  X117.00 Y117.00 Z10.00
Recv: >  X30.00 Y30.00 Z10.00
Recv: echo:busy: processing
Recv: <<< do_blocking_move_to  X30.00 Y30.00 Z10.00
Recv:   current_position= X30.00 Y30.00 Z10.00 : Probe::set_deployed
Recv: deploy: 1
Recv: >>> Probe::run_z_probe  X30.00 Y30.00 Z10.00
Recv: >>> Probe::probe_down_to_z  X30.00 Y30.00 Z10.00
Recv: >>> do_blocking_move_to  X30.00 Y30.00 Z10.00
Recv: >  X30.00 Y30.00 Z-2.20
Recv: <<< do_blocking_move_to  X30.00 Y30.00 Z-2.20
Recv:   current_position= X30.00 Y30.00 Z10.00 : sync_plan_position
Recv: <<< Probe::probe_down_to_z  X30.00 Y30.00 Z10.00
**Recv: FAST Probe fail! - Triggered early.**
Recv: <<< Probe::run_z_probe  X30.00 Y30.00 Z10.00
Recv:   current_position= X30.00 Y30.00 Z10.00 : Probe::set_deployed
Recv: deploy: 0
Recv: >>> do_blocking_move_to  X30.00 Y30.00 Z10.00
Recv: >  X30.00 Y30.00 Z10.00
Recv: <<< do_blocking_move_to  X30.00 Y30.00 Z10.00
**Recv: Error:Probing Failed**
Changing monitoring state from "Operational" to "Error: Probing Failed"
Send: M112
Send: N2 M112*35
Send: N3 M104 T0 S0*34
Send: N4 M140 S0*97
Changing monitoring state from "Error: Probing Failed" to "Offline (Error: Probing Failed)"
Connection closed, closing down monitor
OCTO112: Power off

Ranney1, 2020-10-07:
Update after lot of try and research;

By changing these settings in the Configuration_adv.h can you bypass this error. I cannot trigger an error anymore. What it will do is when failing (the second G29) bypassing it by automaticly triggering an G28 followed by an G29.

/**
 * Repeatedly attempt G29 leveling until it succeeds.
 * Stop after G29_MAX_RETRIES attempts.
 */
**#define G29_RETRY_AND_RECOVER**
#if ENABLED(G29_RETRY_AND_RECOVER)
  **#define G29_MAX_RETRIES 1**
  #define G29_HALT_ON_FAILURE
  /**
   * Specify the GCODE commands that will be executed when leveling succeeds,
   * between attempts, and after the maximum number of retries have been tried.
   */
  #define G29_SUCCESS_COMMANDS "M117 Bed leveling done."
  #define G29_RECOVER_COMMANDS "M117 Probe failed. Rewiping.\nG28\nG12 P0 S12 T0"
  #define G29_FAILURE_COMMANDS "M117 Bed leveling failed.\nG0 Z10\nM300 P25 S880\nM300 P50 S0\nM300 P25 S880\nM300 P50 S0\nM300 P25 S880\nM300 P50 S0\nG4 S1"

#endif

Sebazzz, 2020-10-28:
I think this is fixed in the extui branch.

Sebazzz, 2020-11-01:
No longer happens in extui. I have been bed leveling extensively.

Ready for next release.

closed 2020-11-01 by Sebazzz