CR6Community/Marlin #9: Implement GCODE for Hotend LED

Ranney1, 2020-10-04:
If changed the Configuration_adv.h in firmware to this:
Than is the case light (M355) supported and working

/**
 * M355 Case Light on-off / brightness
 */
#define CASE_LIGHT_ENABLE
#if ENABLED(CASE_LIGHT_ENABLE)
  #define CASE_LIGHT_PIN 6                    // Override the default pin if needed
  #define INVERT_CASE_LIGHT false             // Set true if Case Light is ON when pin is LOW
  #define CASE_LIGHT_DEFAULT_ON false         // Set default power-up state on
  #define CASE_LIGHT_DEFAULT_BRIGHTNESS 255   // Set default power-up brightness (0-255, requires PWM pin)
  //#define CASE_LIGHT_MAX_PWM 128            // Limit pwm
  //#define CASE_LIGHT_MENU                   // Add Case Light options to the LCD menu
  #define CASE_LIGHT_NO_BRIGHTNESS            // Disable brightness control. Enable for non-PWM lighting.
  //#define CASE_LIGHT_USE_NEOPIXEL           // Use NeoPixel LED as case light, requires NEOPIXEL_LED.
  //#if ENABLED(CASE_LIGHT_USE_NEOPIXEL)
  //#define CASE_LIGHT_NEOPIXEL_COLOR { 255, 255, 255, 255 } // { Red, Green, Blue, White }
  #endif

ritchiedc, 2020-10-15:
I have extended this by adding support to update the screen of anyone wants it.

Ranney1, 2020-10-25:
> I have extended this by adding support to update the screen of anyone wants it.

Ahh, great! Can you tell me how you did this? Thanks!

ritchiedc, 2020-10-25:
Attached is a text file detailing the changes I made. It was made before the release today (10/25) so line numbers may be different. I included snippets of code with "c", "+" and "-" in front of lines changed, added or removed. I tried to provide explanations.

[Issue 9 changes.txt](https://github.com/CR6Community/Marlin/files/5435330/Issue.9.changes.txt)

Ranney1, 2020-10-25:
Ahh, thanks! Looking in to it!

Sebazzz, 2020-10-28:
M355 is implemented in extui branch

Sebazzz, 2020-11-01:
Ready for next release.

closed 2020-11-01 by Sebazzz