CR6Community/Marlin #11: Filament runout detection

Ranney1, 2020-10-05:
I've got the filament runout configured like this:

#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define FIL_RUNOUT_ENABLED_DEFAULT true // Enable the sensor on startup. Override with M412 followed by M500.
  #define NUM_RUNOUT_SENSORS   1          // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.
  #define FIL_RUNOUT_STATE     **LOW**       // Pin state indicating that filament is NOT present.
  **#define FIL_RUNOUT_PULLUP**             // Use internal pullup for filament runout pins.
  //#define FIL_RUNOUT_PULLDOWN             // Use internal pulldown for filament runout pins.

  // Set one or more commands to execute on filament runout.
  // (After 'M412 H' Marlin will ask the host to handle the process.)
  #define FILAMENT_RUNOUT_SCRIPT "M25"

  // After a runout is detected, continue printing this length of filament
  // before executing the runout script. Useful for a sensor at the end of
  // a feed tube. Requires 4 bytes SRAM per sensor, plus 4 bytes overhead.
  **#define FILAMENT_RUNOUT_DISTANCE_MM 50**

  #ifdef FILAMENT_RUNOUT_DISTANCE_MM
    // Enable this option to use an encoder disc that toggles the runout pin
    // as the filament moves. (Be sure to set FILAMENT_RUNOUT_DISTANCE_MM
    // large enough to avoid false positives.)
    //#define FILAMENT_MOTION_SENSOR
  #endif
#endif

I know that the filament sensor is not implemented in this firmware. But what i see is that de detection is correct with these settings (GCODE M119) but that the filament runout distance mm (GCODE M412) is always set to 0mm. Whatever you change this value, it is always set to 0mm.  Can this trigger the false positives?

Sebazzz, 2020-11-01:
I think filament runout is working correctly now. Ready for next release (extui).

closed 2020-11-01 by Sebazzz

lottacustoms, 2021-01-26:
Hello,

could you please show your solution :)

Best regards
Marco

Sebazzz, 2021-01-26:
@lottacustoms The filament runout sensor is using the Marlin native implementation from release 4 alpha and higher. If you have an issue, and it is caused by the firmware - not the hardware - please open a new issue.