CR6Community/Marlin #12: Raise Z-Height During On-Screen End-Print Process

formless63, 2020-10-09:
CR-6 SE Specific Issue

### Description

Z does not raise when using the on-screen end-print functionality.

### Steps to Reproduce

1. Begin a print.
2. After printer is laying down plastic, hit end print.

**Expected behavior:** [What you expect to happen]
Expect Z to raise to avoid crashing before homing.
**Actual behavior:** [What actually happens]
Z stays at existing height when print ends and then homes.
#### Additional Information
If on layer one, it appears that the machine stop respecting the bed mesh while homing and drags the nozzle on the glass bed. This is the primary concern that if your un-adjustable bed is higher on the homing side your nozzle may either dig into the glass or into the print and cause damage.
Note that this is the same behavior as the factory firmware and is an issue in that firmware as well.

[image]

formless63, 2020-10-09:
Appears solution may be in config_adv define EVENT_GCODE_SD_ABORT "G0 Z5\nG28XY\nM84"  to raise Z during that process. Thanks Chiefo - will test this tomorrow.

formless63, 2020-10-09:
Ok - compiled with G27 in place of G28XY in that spot.
Just tested. Physical end behavior is perfect. Raises Z up to Z20 or adds 2mm above current position if higher than Z20. Then sends to X0 and Y200-ish (presents the print, like what happens at the end of successful prints)
There is a problem, though. The screen sticks at the "Stop Printing? Confirm Cancel" area and no way to get back to home screen other than power cycle.
I waited two minutes to power cycle to check and confirm temps on nozzle/bed have dropped (that they are shut off) and bed had dropped 20C and nozzle was down close to ambient after a couple minutes. So this works. Creality probably had a conflict with the screen with G27 and used G28XY instead of fixing the screen problem. I'd recommend going with G27 here and power cycling after to protect the printer/bed.

64Spaces, 2020-10-09:
> Thanks - letting @nickacker do the final confirmation because I'm still without CR-6 SE. (I could test it on my BIQU B1 but not sure how representative it would be)

Sebastiaan, thank you for all you do for this community and you still don’t have your printer! I hope it arrives soon. When setup properly and safely, it is a fantastic printer. I hope you get to enjoy it soon.

Sebazzz, 2020-10-09:
I currently have a BIQU B1 (actually for review - BTT tracked me down via the unboxing guide I wrote) and am very happy with that printer. I'm curious how the CR-6 SE will compare :)

closed 2020-10-25 by Sebazzz

DreadKnight, 2020-10-26:
I've ruined my bed because of this bug, as the bed chipped off after removing a print because of some of the scratches. :cry: 
Great printer overall otherwise, I'm glad this is actually fixed now, will try it out soon :+1:

64Spaces, 2020-10-26:
Same for me. I got a big scratch on my first print. I rotated the bed around and avoid that area when slicing. It’s been fine since this update.

provalinf, 2020-10-26:
I was actually looking for a solution for this problem of the nozzle rubbing the bed.
Thanks (even if like many others my brand new bed has already been scratched...)
Now there is indeed this popup problem that is not closed.