CR6Community/Marlin #16: Filament run-out sensor immediately triggered with fw v2.0.0-bugfix-a1

hondrus31, 2020-10-14:
### Description

Same as issue #7 , but with latest release available. After aborting and then starting the print again, it goes on flawlessly.

#### Additional Information

Using start/end gcode found on [r/CR6's wiki](https://www.reddit.com/r/CR6/wiki/index#wiki_start_g-code) , using Cura 4.7.1 with printer profiles downloaded from Cura's github.

hondrus31, 2020-10-15:
Same issue when sending gcode over the usb port via Printrun. Using TeachingTech's generated gcodes.
Again, it triggers the filament runout error only with the first gcode i send, then after aborting once it goes straight forward.

Sebazzz, 2020-10-26:
Can you check if this happens with the latest release and re-open this if it does? You may need to enable filament runout using `M412 S1` first.

closed 2020-10-26 by Sebazzz

hondrus31, 2020-10-26:
Enabled in configuration.h file when building the firmware, using same old gcodes, and it does not give any error with the latest firmware.
Thank you for your work, I wish I could help more :)