CR6Community/Marlin #17: Pause at height

dudzio12, 2020-10-17:
Hi,

Did you manage to get pause at height (M0) running? Can anyone confirm it's working on your newest version?

Thanks for great work 👍

Sebazzz, 2020-10-17:
`EMERGENCY_PARSER` support is enabled, however, continuing the print from the touch screen is currently not supported (I expect that will be implemented some time in the extui branch, together with `ADVANCED_PAUSE_FEATURE`). You would need to issue M108 manually to continue.

artworkvideo, 2020-10-18:
if I remember correctly, here's how I got filament switch to work:

1) In Cura, goto "Extensions"/"PostProcessing"/"Modify G-code" and select "Pause at Height" under "Add a script"
2) change Height setting to Layer (your layer number here)
3) under Method, select BQ(M25)  (this is the key setting)
4) leave park settings at 190,190...(important)
5) set retraction to 200mm...or use a distance you like here
6) set extrude amount to 10mm ..or whatever

when the printer stops, swap filaments, pushing new filament all the way in

press the "Pause" button...WAIT 5 seconds, then press "Resume"...WAIT 10 seconds, the press Print

be sure to remove the "pause at height" steps after final slicing, so they don't end up in your next project 

This may take some practice, but works with the CR-6...thanks to a YouTube clip I can't recall

Sebazzz, 2020-10-28:
Fully implemented on extui branch

Sebazzz, 2020-11-01:
Ready for next release.

closed 2020-11-01 by Sebazzz