CR6Community/Marlin #18: [BUG] Auto bed leveling freezing

giorgiomanieri, 2020-10-25:
### Bug Description

When I start the auto bed leveling after the home position, the nozzle move to first probing and immediately turn back in origin

### My Configurations

Community firmware v1.0.3.6

### Steps to Reproduce

1. Start auto bed leveling from the screen

**Expected behavior:** 

Probing from 1st to 16th point

**Actual behavior:** [What actually happens]

Nozzle stops on first probing

(https://youtu.be/klTeJAVzQS0)

Sebazzz, 2020-10-25:
Please try to execute G29 from Pronterface and collect logs by activating `M111 S247` first.

giorgiomanieri, 2020-10-25:
> Please try to execute G29 from Pronterface and collect logs by activating `M111 S247` first.

Where I find the logs then?

Sebazzz, 2020-10-25:
In the pronterface console

Met vriendelijke groet,
Sebastiaan Dammann

giorgiomanieri, 2020-10-25:
> In the pronterface console Met vriendelijke groet, Sebastiaan Dammann
> […](#)

I'm sorry, I don't read any log there

Sebazzz, 2020-10-25:
[firmware-20201025-205250.zip](https://github.com/CR6Community/Marlin/files/5435830/firmware-20201025-205250.zip)

Please try flashing this firmware and try again:

1. Verify flashing with `M115`.
2. Enable debug logging with `M111 S247`
3. Execute G28
4. Execute G29

giorgiomanieri, 2020-10-25:
Recv: Error:Probing Failed
Changing monitoring state from "Operational" to "Error: Probing Failed"
Send: M112
Send: N2 M112*35
Send: N3 M104 T0 S0*34
Send: N4 M140 S0*97
Changing monitoring state from "Error: Probing Failed" to "Offline (Error: Probing Failed)"
Connection closed, closing down monitor

Sebazzz, 2020-10-25:
Is this the only output you get? I'd expect much more output. Can you post the full output of everything in Pronterface, following the instructions? :)

nickacker, 2020-10-25:
More information is required about your current firmware situation @giorgiomanieri. Are you still using 1036? Is this the 1036 build I released or one you've built from the source? Have you attempted to update the firmware to the latest release from this repo?

In the youtube video you linked above, it would appear as though your strain gauge LED is triggered just before it hits the X endstop. The fact that it turns on properly during the home movement tells me that its likely something to do with the bowden/data cable setup you have on it. My guess is that the cable/bowden is pulling the assembly up as it reaches the endstop and therefor, its not able to finish or even start because its already triggered before the optical sensor even.

Remove the data cable saver print, as well as all of the clips youre using to secure the data cable to the bowden tube and try again. Pay very close attention to the blue LED thats under the hotend shroud as it should remain off unless the nozzle is actually pushing on something. If remove the 3d prints from the cables sorts the issue out, then you can certainly go back to using them, but just turn the potentiometer up slightly so accomodate for the uppull on the assembly.

nickacker, 2020-10-25:
After careful review of the video posted above in the OP. @0:20 seconds, the strain gauge probe is triggered before the X endstop, remains on during the first attempted point, and then turns back off when it resets back to home middle position, indicating that the gauge is working properly. This is expected behavior from a failed first probe and is caused by the strain gauge being activated already before the optical switch.

User to remove the additional clips supporting the data cable and increase resistance to the potentiometer as this is not firmware related.

closed 2020-10-25 by nickacker

giorgiomanieri, 2020-10-26:
Ok. I've tried. It works for now.
Thanks