CR6Community/Marlin #19: Change in default for runout sensor is a breaking change, makes old slices useless

joesanford, 2020-10-25:
### Description

In [Community Release 3](https://github.com/CR6Community/Marlin/releases/tag/v2.0.7.1-cr6-community-release-3) the default for the runout sensor was changed from enabled to disabled. For those of us with functioning runout sensors, this is a breaking change if we were to upgrade to the new firmware, or we have to re-slice any old models we already have.

This is a proposal to revert https://github.com/CR6Community/Marlin/commit/6d5bdd5558492f78f026013d6bf7905bd330a314, and update documentation fo those with filament runout sensors to add the correct gcode to their slicer configuration.

Sebazzz, 2020-10-25:
> For those of us with functioning runout sensors, this is a breaking change if we were to upgrade to the new firmware, or we have to re-slice any old models we already have.

This is a breaking change only if updating from Creality firmware. 

By creating a gcode file with:

```
M412 S1
M500
```

It is easily and "permanently" enabled. 

It is currently disabled by default because some members (@nickacker among others) had issues with filament runout and needs more testing before I want to enable by default.

joesanford, 2020-10-25:
Yes, however this leaves me (and all of the others with working runout sensors) in quite the predicament. I have ~25 models sliced and waiting on my Octoprint Pi for me to print. I upgraded my firmware this morning. All of those models are either going to need to be resliced, and I'm also trying to figure out if downgrading the firmware is now possible.

I understand the motive and reasoning behind the change, but what I don't get is the motive for making a change that breaks backwards compatibility with code previously sliced for this printer. Especially when you also say this is something that will eventually change.

IMO those with broken sensors should be doing a workaround in the meantime, instead of disabling functionality for those with functioning hardware. This is a very backwards and confusing approach to a problem like this.

nickacker, 2020-10-25:
As the development of the community firmware continues, we will attempt to make the best efforts to ensure backwards compatibility with previous slices. In this case we found the majority of use cases simply dont need the runout detection for now and as such until it can be detected properly and resumed properly, we've simply turned it off on the startup of the machine. 

This is untested(but should work just fine), using pronterface or other machine interface, you can send M412 S1 to turn the setting back on, and then M500 to save it permanently which will restore your previous slice run out detection and future as well. 

If you dont have a machine interface. You can also create a basic text file with these two lines. Then rename it to a .gcode file and have the printer 'print' this file which will renable and save your runout state.

closed 2020-10-25 by nickacker

joesanford, 2020-10-25:
Again, the point that is being missed is that you are disabling a piece of hardware, that by default is enabled, and made the workaround instead for people with functioning hardware. And your proposed fix isn't even mentioned anywhere. People on reddit and Facebook

It makes no sense to break backwards compatibility for the sake of a workaround, especially when it's a feature of the hardware. 

Just because _you_ are facing an issue does not mean it should be merged into mainline code, especially given the above. I don't see how you can call this "community" firmware when you're tailoring it to your needs.

joesanford, 2020-10-26:
Thank you for providing a solution until runout sensor hardware is fixed for everyone.