CR6Community/Marlin #20: [FR] Add M300 (Play Tone) support

hondrus31, 2020-10-26:
### Description

Since this printer is very quiet, adding a little sound at the end of the gcode it's a nice touch.
I tried building the firmware with _#define SPEAKER_ enabled in _configuration.h_ file, but when sending an M300 to the printer, the console outputs an "unknown command", with no sound played at all.

While trying to search for some fixes among the files, the only thing I was able to understand is that there is no beeper reference in _pins_CREALITY_V452.h_ file, if that can help.

Sebazzz, 2020-10-26:
Yes, I'd like this as well. My BIQU B1 has a speaker as well and I added end-gcode when a print is finished.

Will pick this up as part of the extui porting efforts.

martinbogo, 2020-10-28:
Is there a speaker in the CR6 SE?

Sebazzz, 2020-10-28:
Fully implemented on extui branch

Sebazzz, 2020-10-28:
> 
> 
> Is there a speaker in the CR6 SE?

Yes, and no. There is a speaker in the screen and we can ask the screen to beep for a number of time units (8ms minimum). We can't change the pitch though.

Sebazzz, 2020-11-01:
Ready for next release.

closed 2020-11-01 by Sebazzz