CR6Community/Marlin #21: [BUG] With current (10/26/2020) community firmware and screen firmware, sometimes stuck trying to bed level due to 0C temp goal?

dwendooley, 2020-10-26:
On occasion, it appears the automatic bed leveling does not work because it pauses waiting for the hot end to reach the desired temp of 0. After a reboot it usually comes back with an expected temp of 120, but not always. 
Menu is not disabled, can back out of the wait, but only a reboot seems to clear the issue.  I will attempt to replicate and see if it's in response to any button presses in particular.

Doridian, 2020-10-26:
Same here. I thought it was an issue because I just flashed and after a reboot it was gone.
When this happens, not only does it show a target of 0C, but it also shows no numbers in the field below (where it'd show all the points 1-16 to indicate progress)

Linnie73, 2020-10-27:
Same here. After flashing the firmware, I had the SD card still in the printer. Bed leveling command through the lcd would not work because the temperature was set to 0 degrees. 
Powercycled the printer, the SD card was still inserted. Same result. Powered off, took SD card out, and retried bed leveling. This time it started, heated up the nozzle to 120, homed and then moved to the center of the bed and then the system froze. Back button not responsive. Power cycled and tried again and it has a 0 degree target. After another power cycle, it started the procedure correctly. Not sure if the presence or not of the SD card had anything to do with it since the last 3 power cycles with different results did not have the SD card present.

64Spaces, 2020-10-27:
I had this happen with the previous version of the community firmware. Nothing scientific, but I believe it needed a Factory Reset, after firmware flash. I’ll keep an eye on it.

Doridian, 2020-10-27:
As a work-around atm (possibly, I haven't had the bug happen again, but it works normally), you can send the following GCode to initialize levelling:
```
M109 R120 ; Set nozzle to 120C and wait
G28 ; Home
G29 ; Mesh Levelling
M500 ; Save to EEPROM
M104 S0 ; Set nozzle to 0C (off)
```

Sebazzz, 2020-10-28:
Fixed on extui branch

Sebazzz, 2020-11-01:
Ready for next release.

closed 2020-11-01 by Sebazzz