CR6Community/Marlin #23: [FR] Arc Support

ZanSchaffer, 2020-10-26:
### Description

I would like to use https://github.com/FormerLurker/ArcWelderPlugin, but it requires ARC support. Is that something that can be implemented?

### Feature Workflow
Ability to process G2 and G3 commands.

```
Send: G2
Recv: echo:Unknown command: "G2"
```

ZanSchaffer, 2020-10-27:
Tried it. Seems to be working without issue out of the box.

Sebazzz, 2020-10-27:
Still keeping this open though - we still need to implement this in the default config we ship :)

Sebazzz, 2020-10-28:
@Nushio or @nickacker you already committed something for this, no?

Nushio, 2020-10-28:
I basically turned the feature on. I've tested this a bit (about 8 different prints, from sdcard and octoprint) and so far, nothing's caught fire.

Sebazzz, 2020-10-28:
My question was whether you committed it 😁

Met vriendelijke groet,
Sebastiaan Dammann

Sebazzz, 2020-10-29:
Fixed on main branch and extui branch.

Sebazzz, 2020-11-01:
Ready for next release.

closed 2020-11-01 by Sebazzz