CR6Community/Marlin #25: [BUG] Lock up after stopped print

crysalis010, 2020-10-28:
When stopping a print from touchscreen, the print successfully stops, but screen locks up at the confirmation page. Must reboot in order to start over.

Doridian, 2020-10-28:
I am 99.9% sure this is because of https://github.com/CR6Community/Marlin/commit/1ed291cede7b6e5df114b97fce52622b4b929baa
I remember reading this as a "known issue", but can't find it documented anymore.
It fixes one bug (possible physical damage to your printer) with another (hanging on the abort screen)

Sebazzz, 2020-10-28:
I don't see how that could cause it. There is no real difference between executing one gcode or the other.

As mentioned in the release notes, we need logging to diagnose this. Please provide some 👍

crysalis010, 2020-10-28:
I’m happy to provide logs if told where to find them/get them.

starfuryth, 2020-10-28:
+1
Updated firmware yesterday, ran straight into this issue.

martinbogo, 2020-10-28:
I have also reproduced this issue.   If you start a print, then stop it from the touchscreen, the screen locks up at the confirmation page and accepts no further input.

Doridian, 2020-10-28:
I know I read it somewhere: https://github.com/CR6Community/Marlin/pull/13
`This does introduce a problem of the screen freezing on the "Stop Printing?" screen`

Sebazzz, 2020-10-28:
Fixed on extui branch

Sebazzz, 2020-11-01:
Ready for next release.

closed 2020-11-01 by Sebazzz