CR6Community/Marlin #27: [BUG] Support print progress display

ZanSchaffer, 2020-10-30:
### Bug Description

I am using https://plugins.octoprint.org/plugins/DisplayLayerProgress/ and it doesn't show print progress on the LCD during print.

### My Configurations

Community release 3 + ARC support

### Steps to Reproduce

1. Open terminal
2. M117 xxx% L=xxx/xxx H=xxx.xx/xxx.xx
3. See if anything happens on LCD

**Expected behavior:** Some kind of print progress display. 

**Actual behavior:** Printer ignores command.

ZanSchaffer, 2020-10-30:
May be a feature request?

Sebazzz, 2020-10-30:
Indeed a feature request, M117 was never implemented.

Implemented in extui branch, though currently we only display M117 on the print screen and not on the home screen.
It is different in the CR-6 SE touch screen opposed to the Marlin native display, in the sense that on Marlin the "home/info" screen is the print screen, where the CR-6 SE has a "main menu" and a print screen.

Sebazzz, 2020-11-01:
Implemented in ddc4e4f1518ae4cb0921ed6079d7fd3241432d54 - ready for next release.
Should be used in conjunction with M75 and M77.

closed 2020-11-01 by Sebazzz