CR6Community/Marlin #30: [BUG] libmaple/usart.h: No such file or directory

Ajtak, 2020-11-01:
If I tried build in Arduino IDE, I'm getting this error:

```
sketch\src\lcd\dwin\cr6\touch_lcd.cpp:33:10: fatal error: libmaple/usart.h: No such file or directory
 #include <libmaple/usart.h>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
```

How to fix it?
Thanks

Sebazzz, 2020-11-01:
Not use Arduino IDE. Use Visual Studio Code with Platform.io plugin. 

Which branch are you building?

joesanford, 2020-11-01:
How are you building it, and what board are you targeting? You may be building for more than one type of board.

VSCode/Platformio works, but Marlin should support Arduino IDE.

Ajtak, 2020-11-01:
I'm building default branch (**creality-cr6-merge-attempt**) for **Arduino Mega or Mega 2560** board

Sebazzz, 2020-11-01:
That's probably the issue here. Can you please try VSCode?

Doridian, 2020-11-01:
@Ajtak The printer uses an STM32, not an 8-bit Arduino, so some libraries might be broken just from that. I don't know if the Arduino IDE can compile Marlin for STM32, but you have to at least install the board plugin for them and set the correct chip.

hondrus31, 2020-11-02:
Tried many times with Arduino IDE with no success. VScode + Platform.io + Auto Build Marlin worked from the first time. 
Consider following the guide on official Marlin site [here](https://marlinfw.org/docs/basics/auto_build_marlin.html).

joesanford, 2020-11-04:
Apologies, half of my comment didn't post - was going to add that since everyone on this fork is a VSCode user (AFAIK), it may be useful asking in the Marlin project this is forked from: https://github.com/MarlinFirmware/Marlin

Sebazzz, 2020-11-05:
Closing this as this is probably an upstream issue.

closed 2020-11-05 by Sebazzz