CR6Community/Marlin #32: [BUG] (Auto Level hangs)

jacquesbotha, 2020-11-02:
Under specific conditions, the Auto level function will hang the printer (make it non-functional).  Needing a reboot to function again.

**Condition:**
Strain gauge very sensitive. [Worked fine with Creality 1.3 version]
Blue touch LED will blink very briefly during moving when tube is extended. [But not continuously on]

ABL centers, and touches. [ok]
Then moves to position 1 [ok]
As it moves down, the blue touch LED flickers briefly [but not on perm - due to high sensitivity]
ABL aborts, and moves to the center.
Printer hangs, needing a reboot to function again.

**Temp Fix:**
I re-adjusted the sensitivity to cause it not to flicker briefly while in pos 1 [tube is stretched and put pressure on gauge]
This makes the gauge less sensitive.

**Proposed software fix:**
Build in some de-bouncing function for the gauge.  Maybe add a sensitivity counter / threshold where it has to be on for ? micro seconds before detecting a touch.
This will make it very sensitive, and also make it work with the strain of the tube loaded on the gauge.

Sebazzz, 2020-11-02:
Which release are you on? Is this the same as  #21?

jacquesbotha, 2020-11-02:
> Which release are you on? Is this the same as #21?

Community firmware release 3 based on Marlin v2.0.7.1
Nope - not related - the bed heats up, but the problem is related to no de-bouncing of the strain gauge.

Sebazzz, 2020-11-02:
Your comments regarding the strain gauge made me think, so I went ahead, enabled DIRECT_PIN_CONTROL to directly talk to the strain gauge. 

When COM_PIN is set to 0 it turns off reporting of the Z-endstop. To test this, move the hot-end so it is triggered (blue led), then set COM_PIN to LOW (via M42 P5 S0). The blue LED will turn off.

M119 also says that z_min = open. When you then turn on COM_PIN to 1, the Z-endstop is still not triggered.

If you then apply even more force to the strain gauge, it is then triggering again. So the combination of COM_PIN 0, a delay, and COM_PIN 1 is essentially the act of pressing the "tare" or "zero" key on your kitchen scale. It is self-calibrating (when the Marlin tells it to) which begs the question what that potmeter is even doing there. Maybe all it defines is the threshold. You tare it via firmware, but the threshold (how many "grams" it takes for it to trigger) is set via the potmeter.

Sebazzz, 2020-11-02:
I've done some improvements in eede631f.

closed 2020-11-05 by Sebazzz