CR6Community/Marlin #33: [BUG] While pre-heating, ABL has no effect on adjusting the temp down to 60

jacquesbotha, 2020-11-02:
I was wanting to pre-heat the printer, and do a ABL (To quickly get the bed up to 60deg)

**To replicate**
On boot, navigate to pre-heat PLA
Temperatures were set to ?/200 and ?/60 
When bed reached 60, I selected Auto Bed Level
Nozel temperature was set to ?/120 [at the time, the nozel temp was 132] (bed temp was set to 0)
During ABL, the nozel temperature kept climbing until it reached 200 (although the nozel temp was set to 120) (bed temp dropped fine)

**Potential software issue:**
During ABL, the temp is displayed on the screen as 120, but somehow the set nozel temp is still stuck on 200 from pre-heat.
Only when it reached 200, then it drops.
- Nice to have: don't let ABL adjust the bed temp. (I would like to ABL when the bed is at it's working temp of 60)

Sebazzz, 2020-11-02:
Which branch?

jacquesbotha, 2020-11-02:
> Which branch?

Community firmware release 3 based on Marlin v2.0.7.1

Sebazzz, 2020-11-02:
This does not appear to happen on the extui branch anymore.

jacquesbotha, 2020-11-03:
> This does not appear to happen on the extui branch anymore.

Perfect!  Thanks

Sebazzz, 2020-12-23:
[Creality v4.5.2 motherboard firmware-crealityv452-20201223-002158.zip](https://github.com/CR6Community/Marlin/files/5732408/firmware-crealityv452-20201223-002158.zip)
[Creality v4.5.3 motherboard firmware-crealityv453-20201223-002246.zip](https://github.com/CR6Community/Marlin/files/5732406/firmware-crealityv453-20201223-002246.zip)
[Touch screen firmware CR-6-touchscreen-2020-12-23.zip](https://github.com/CR6Community/Marlin/files/5732436/CR-6-touchscreen-2020-12-23.zip)

Can you please retry this with one of these firmware files, depending on the motherboard version you have. If you don't know or got a printer from November 2020 or earlier, you'Il most likely have the Creality v4.5.2 motherboard. 

Assuming we got it fixed I will close this issue now and consider it fixed. If it isn't fixed please provide:
- Logs captured via Pronterface/Octoprint/PuTTY with `M111 S247` entered for debug logging
- Pictures or a video as appropriate

Also, if it is fixed, I still would like to have a confirmation though😀

closed 2020-12-23 by Sebazzz