CR6Community/Marlin #34: Cancel Print - If the print doesnt stick you press cancel which works. But then locks the screen and requires restart.

Benny-H, 2020-11-05:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Sebazzz, 2020-11-05:
Which branch? Did this happen on stock firmware as well?

Sebazzz, 2020-11-05:
Do you have any logs? (M111 gcode + serial monitor)

Benny-H, 2020-11-05:
Sorry im not that good with the programming side. Basically im having
issues with the first layer sticking and the community firmware was said to
help as it improves the resolution of the tuning to 0.01 from 0.05.

I tried to mess with it but it didnt stick so i press cancel and get a
"confirm" screen same as normal. I then press confirm and the printer does
stop but the screen doesnt return to the main menu as it did before. It
stays on the confirm screen until i shut it down. When i restart it tries
to continue the print.

The other issue that im having is that the Z offset is only 0.01 on the
levelling screen. What i want to do is change the Z offset while printing
using the "tune" screen which the software allows but its only 0.05 still.

Ben

Benny-H, 2020-11-05:
No the stock firmware returns to the main screen when you confirm cancel.

Sebazzz, 2020-11-05:
Fixed in issue #25

Benny-H, 2020-11-05:
Thanks for this, I assume I have downloaded an old version then.

Any chance you can send me the latest version link?

I assume this also addresses the "cant change the Z height by 0.01 (or
display to 0.01) during printing" as well?

It would really help me out if it did!

Ben

Sebazzz, 2020-11-05:
It's not released yet. For the 0.01 you may need to flash the [v1.0.0-a1](https://github.com/CR6Community/CR-6-touchscreen/releases/tag/v1.0.0-a1) touch screen firmware.

closed 2020-11-05 by Sebazzz

Benny-H, 2020-11-05:
I have been trying to work out how to actually flash the screen and i think
i need a smaller SD card and have to take the screen cover off. Seems crazy!

One other thing, if possible can you please add to the running display
which layer and total layers display? It has the time and progress bar but
i would rather see the layer number.

Ben

Benny-H, 2020-11-05:
Sorry to bother you again but i had a print fail from the SD card of the
printer, so one i assume would work perfectly every time and i wondered if
it was the firmware update to yours that caused the issue.

I tried to flash back to the Creality software from their website but its
now made my printer chinese only! There is no option to change back! Any
ideas why?

It didnt happen when i last flashed this same software although im not
totally sure i got the last one from their website but i am pretty sure i
did.

I tried the print again and it failed the same way so its not your firmware
so im going back to yours but i hope that its not still in Chinese!

Ben

Sebazzz, 2020-11-05:
> One other thing, if possible can you please add to the running display which layer and total layers display? It has the time and progress bar but i would rather see the layer number.

I'm not sure Marlin knows the layer number, but we can show the position of the Z-axis of course.

> Sorry to bother you again but i had a print fail from the SD card of the printer, so one i assume would work perfectly every time and i wondered if it was the firmware update to yours that caused the issue. I tried the print again and it failed the same way so its not your firmware so im going back to yours but i hope that its not still in Chinese!

Language settings are controlled from the LCD screen, but as you've found out the Creality CR-6 SD card is unreliable. Use a new SD card.

Benny-H, 2020-11-06:
Got another one for you. The abl has stopped working now. It goes to the first corner then returns to the centre before actually doing the Touch. The screen is locked up. Requires restart. Cheers,Benjamin HeynenBenjamin.heynen@gmail.com07 494 686663-------- Original message --------From: Sebastiaan Dammann <notifications@github.com>Date: Thu, 5 Nov 2020, 23:00To: CR6Community/Marlin <Marlin@noreply.github.com>Cc: Benny-H <benjamin.heynen@gmail.com>, Author <author@noreply.github.com>Subject: Re: [CR6Community/Marlin] Cancel Print - If the print doesnt stick you press cancel which works. But then locks the screen and requires restart. (#34)

One other thing, if possible can you please add to the running display which layer and total layers display? It has the time and progress bar but i would rather see the layer number.

I'm not sure Marlin knows the layer number, but we can show the position of the Z-axis of course.

Sorry to bother you again but i had a print fail from the SD card of the printer, so one i assume would work perfectly every time and i wondered if it was the firmware update to yours that caused the issue. I tried the print again and it failed the same way so its not your firmware so im going back to yours but i hope that its not still in Chinese!

Language settings are controlled from the LCD screen, but as you've found out the Creality CR-6 SD card is unreliable. Use a new SD card.

Benny-H, 2020-11-06:
Guess I'm a good beta tester because I have tried to go back to any of the creality software and since doing the community the machine only displays Chinese on the creality software. I can get to the setting to change it but that button does not respond. When I got the machine I went from what it shipped with to the latest so it's only an issue going back to creality from community. Having a great time here!! Also found that my extruder is jamming because I tightened a loose bolt and I'm not sure why, but every time I bed level the z offset is different. Now I am making sure that the sensor trips when you manually change z to zero. It has been plus it minus 0.1 back to back! Cheers,Benjamin HeynenBenjamin.heynen@gmail.com07 494 686663-------- Original message --------From: Ben Heynen <benjaminheynen@gmail.com>Date: Fri, 6 Nov 2020, 10:54To: CR6Community/Marlin <reply@reply.github.com>, Sebastiaan Dammann <notifications@github.com>, CR6Community/Marlin <Marlin@noreply.github.com>Cc: Benny-H <benjamin.heynen@gmail.com>, Author <author@noreply.github.com>Subject: Re:[CR6Community/Marlin] Cancel Print - If the print doesnt stick you press cancel which works. But then locks the screen and requires restart. (#34)Got another one for you. The abl has stopped working now. It goes to the first corner then returns to the centre before actually doing the Touch. The screen is locked up. Requires restart. Cheers,Benjamin HeynenBenjamin.heynen@gmail.com07 494 686663-------- Original message --------From: Sebastiaan Dammann <notifications@github.com>Date: Thu, 5 Nov 2020, 23:00To: CR6Community/Marlin <Marlin@noreply.github.com>Cc: Benny-H <benjamin.heynen@gmail.com>, Author <author@noreply.github.com>Subject: Re: [CR6Community/Marlin] Cancel Print - If the print doesnt stick you press cancel which works. But then locks the screen and requires restart. (#34)

One other thing, if possible can you please add to the running display which layer and total layers display? It has the time and progress bar but i would rather see the layer number.

I'm not sure Marlin knows the layer number, but we can show the position of the Z-axis of course.

Sorry to bother you again but i had a print fail from the SD card of the printer, so one i assume would work perfectly every time and i wondered if it was the firmware update to yours that caused the issue. I tried the print again and it failed the same way so its not your firmware so im going back to yours but i hope that its not still in Chinese!

Language settings are controlled from the LCD screen, but as you've found out the Creality CR-6 SD card is unreliable. Use a new SD card.

Sebazzz, 2020-11-06:
> Got another one for you. The abl has stopped working now.

Please check issue #18 

> Guess I'm a good beta tester because I have tried to go back to any of the creality software and since doing the community the machine only displays Chinese on the creality software. 

You need to flash the original screen firmware for that.

> Also found that my extruder is jamming because I tightened a loose bolt and I'm not sure why

I don't see how we could have caused that.

Benny-H, 2020-11-06:
Hi again! No I was saying I was having issues and love what you guys are doing and appreciate it! I have not done any screen flashing as I don't have a small sd card. How can firmware flashing have changed the screen stuff? I'll wait for your next release and get an sd card and then do both the firmware and your screens together. I'll bash on with the Chinese version for now. I also noted that on your firmware often registers a button press with a beep but doesn't actually do anything. I found this on the manual feed button as well as randomly on the z height setting. Just for my own knowledge. Does the z height setting in the tube menu when printing change anything other than the first layer offset? I was wondering if you can tube layer heights with that setting while running. Please forgive my noobness! Cheers,Benjamin HeynenBenjamin.heynen@gmail.com07 494 686663-------- Original message --------From: Sebastiaan Dammann <notifications@github.com>Date: Fri, 6 Nov 2020, 13:17To: CR6Community/Marlin <Marlin@noreply.github.com>Cc: Benny-H <benjamin.heynen@gmail.com>, Author <author@noreply.github.com>Subject: Re: [CR6Community/Marlin] Cancel Print - If the print doesnt stick you press cancel which works. But then locks the screen and requires restart. (#34)

Got another one for you. The abl has stopped working now.

Please check issue #18

Guess I'm a good beta tester because I have tried to go back to any of the creality software and since doing the community the machine only displays Chinese on the creality software.

You need to flash the original screen firmware for that.

Also found that my extruder is jamming because I tightened a loose bolt and I'm not sure why

I don't see how we could have caused that.