CR6Community/Marlin #35: Z offset on screen - Change partially worked but not during print.

Benny-H, 2020-11-05:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [Change Z offset works well in the levelling page although the steps are not clear - Sometimes its 0.005 and others 0.01 but display is only to 0.00]
2. [Change Z offset during printing is still in 0.05 steps (so 4 steps to bed contact)]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Sebazzz, 2020-11-05:
Which branch? Did this happen on stock firmware as well? I expect it does, because we did not change this code.

SippieCup, 2020-11-08:
This does happen in stock firmware, It also has some rounding error (even though its only addition/subtraction) which causes the display to show 0.19mm instead of 0.2. This might be related.

ZanSchaffer, 2020-11-08:
Alpha 4 hotfix 2
https://1drv.ms/v/s!Ai7k4vIftfUfhYQGPUQUzMiOvLivAw?e=WjhTwq

may be related to this one https://github.com/CR6Community/Marlin/issues/38

Sebazzz, 2020-11-08:
Due to a faulty motherboard, I'm not able to test it myself and have one hand tied behind my back for that reason, but if you have the time, please follow these steps:

1. Reset to factory settings *or* use [M851](https://marlinfw.org/docs/gcode/M851.html) to reset the probe offset to a good value like Z0.25 and save with M500.
2. Open your serial monitor and start a random print.
3. Adjust Z-offset via the display.
4. Upload the output from the serial monitor annotated with when you adjusted the Z-offset

ZanSchaffer, 2020-11-08:
@Sebazzz serial monitor is the terminal tab in Octoprint?

Sebazzz, 2020-11-08:
> 
> 
> @Sebazzz serial monitor is the terminal tab in Octoprint?

Yes, or pronterface, or PuTTY :)

ZanSchaffer, 2020-11-08:
@Sebazzz 
[z_offset_print.txt](https://github.com/CR6Community/Marlin/files/5507063/z_offset_print.txt)
Noticed a couple of weird lines:
`Recv: - Absolute: 0.24- Existing: 0.25 - Difference: -0.01 - Steps: -5;
Recv: - Absolute: 0.22- Existing: 0.24 - Difference: -0.02 - Steps: -7;
Recv: ok`

Sebazzz, 2020-11-08:
Yes, that is the logging I've added here: https://github.com/CR6Community/Marlin/commit/583a7f25186ded1c05cbd25b3d0ba1480a70f2de#diff-495337e302ee775c3b5d09853c7ecd9d574897d9777eced132cd9131c9ab8987R829

Do the statements logged match what you did on the display?

ZanSchaffer, 2020-11-08:
> Do the statements logged match what you did on the display?
They do I pressed down a couple of times until I have seen some 0.2 jumps.

ZanSchaffer, 2020-11-08:
Also maybe it is related. I use BedVisualiser plugin to level the bed before my print sessions and the first leveling (within a power cycle or reset defaults - don't know which one yet) after after alpha 4 (haven't noticed before) turns out like this.

[image]

JohannesBertens, 2020-11-13:
Also having this, a bit of hit/miss on first layer!

Seem to work consistantly for in the "level" menu, but that is not accessible during first layer print.

Thinkersbluff, 2020-11-14:
> Seem to work consistantly for in the "level" menu, but that is not accessible during first layer print.

@JohannesBertens - Please clarify/elaborate.  What is "not accessible"?  
e.g. I there is a Z Offset adjustment function available on the "Tune" menu, during printing, but it seems to still adjust Z Offset in increments of 0.05mm, not 0.01.  Is that what you mean? or something else?

JohannesBertens, 2020-11-14:
Ah, yes,that is it. In the tune menu it's only with 0.05 increments (or
sometimes 0.04??)

I'll make a movie tomorrow.

Thinkersbluff, 2020-11-14:
> [z_offset_print.txt](https://github.com/CR6Community/Marlin/files/5507063/z_offset_print.txt)
> Noticed a couple of weird lines:

@Sebazzz - I notice that the software is not consistent about how many steps it takes, to move Z by -0.01mm, in that sample.

Recv: - Absolute:   0.24- Existing: 0.25 - Difference: -0.01 - Steps: -5;
Recv: - Absolute: 0.21- Existing: 0.22 -   Difference: -0.01 - Steps: -5;
Recv: - Absolute: 0.18- Existing: 0.19 -   Difference: -0.01 - Steps: -4;
Recv: - Absolute: 0.17- Existing: 0.18 -   Difference: -0.01 - Steps: -4;
Recv: - Absolute: 0.12- Existing: 0.13 -   Difference: -0.01 - Steps: -4;
Recv: - Absolute: 0.11- Existing: 0.12 -   Difference: -0.01 - Steps: -4;
Recv: - Absolute: 0.10- Existing: 0.11 -   Difference: -0.01 - Steps: -4;

I imagine this is because the number of significant figures used by the software exceeds the number of significant figures being displayed.  I realize it is probably a different part of the software that handles "Z-Hop on Retraction", but if the calculations are done the same way it looks like this kind of "rounding error" could accumulate into visible artifacts, during long prints  

I don't know how to code, unfortunately, so I can't just figure this out for myself, but I thought I would ask, in case it triggers a helpful thought.

Sebazzz, 2020-11-14:
I know how to code, but have no working printer. Would you be able to do a few test for me if I can get a build ready?

Thinkersbluff, 2020-11-14:
> Would you be able to do a few test for me if I can get a build ready?

Absolutely.  Happy to.

Sebazzz, 2020-11-14:
[firmware-20201114-191401.zip](https://github.com/CR6Community/Marlin/files/5541259/firmware-20201114-191401.zip)

Please try this.

Thinkersbluff, 2020-11-14:
I am back and starting test.  If you find this before I finish, please advise if you would like specific information and specific procedures to be followed.

Sebazzz, 2020-11-17:
I think that if your Z-offset is already wrong (so not a whole 2-place decimal number) it may still keep flipping.

Thinkersbluff, 2020-11-18:
> I think that if your Z-offset is already wrong (so not a whole 2-place decimal number) it may still keep flipping.

@Sebazzz - sorry, I am out of context and don't understand what you mean.  I did notice today, though, that M503 returned my Z offset as 0.19, when the displayed value was 0.18, so there certainly do seem to be multiple variables representing the same parameter, within the code.

Sebazzz, 2020-11-18:
@Thinkersbluff No there is certainly just one variable. I meant if the variable is near the rounding point the DWIN touch screen might round it one way for presentation and Marlin another way for pr.esentation. So, I bet that if you set your Z offset to, let's say 0.20 in gcode and save to EEPROM it doesn't happen anymore.

Thinkersbluff, 2020-11-18:
> @Thinkersbluff No there is certainly just one variable. I meant if the variable is near the rounding point the DWIN touch screen might round it one way for presentation and Marlin another way for [...]

@Sebazzz I can confirm that M851 Z0.18 did force agreement between M503 and the display.  
I can not explain how they got out of sync in the first place, though, so I don't trust they will stay in harmony. 
Maybe the fact that I am presently not using the version of the firmware you attached above to fix this bug (I am back on the extui branch from 3 days ago) would have allowed me to get out of synch? It nags at my tired brain that Marlin may work internally to more significant figures than the rest of us.  In my blissful non-programmer's innocence, I keep wondering why not force the system to round everything off the same way when it is calculated and before it is stored/distributed/used.  Printing is a game of microns. Adhesion requires precision and repeatability. I wish I could trust the displayed values to be the ones my printer is actually using and not have to double-check with M503 to catch and correct discrepancies.  (Not being critical, just explaining my thoughts...)

Sebazzz, 2020-11-18:
> I can not explain how they got out of sync in the first place, though, so I don't trust they will stay in harmony.

Because I don't trust DWIN either, I just check if the incoming value is lower or higher than the current one and substract or add 0.1mm 😉

dryphi, 2020-11-23:
Could it be a different character type? Not sure how the screen and mainboard interact, but I've run into problems in Arduino where the character type was too small to store the range of values. Fixed it by changing an "int" to a "double", for instance. Just an idea, I'm completely new to this project.

Sebazzz, 2020-11-23:
Whatever it is, it is resolved with the firmware file above - we have now one source of truth: the firmware on the motherboard (Marlin).

dryphi, 2020-11-23:
I guess I was referring to #38 which is the problem I'm having. Thought they were the same thing.

skeaton250, 2020-11-23:
> Whatever it is, it is resolved with the firmware file above - we have now one source of truth: the firmware on the motherboard (Marlin).

Will you be posting the source code for this?

Sebazzz, 2020-11-23:
> 
> 
> > Whatever it is, it is resolved with the firmware file above - we have now one source of truth: the firmware on the motherboard (Marlin).
> 
> Will you be posting the source code for this?

It is pushed to git? https://github.com/CR6Community/Marlin/issues/48#issuecomment-729204369

Sebazzz, 2020-12-10:
Should be fixed in latest sources.

closed 2020-12-10 by Sebazzz