CR6Community/Marlin #38: [BUG] Z offset not consistent

ZanSchaffer, 2020-11-07:
### Bug Description

If I press the up and down buttons on leveling screen it sometimes increments  for 0.02 instead of 0.01. For example I can't hit 0.15 offset because it goes from 0.16 directly to 0.14.

### My Configurations

alpha 4 release & the newest monitor update

### Steps to Reproduce

1. Start at 0.20
2. Go down to 0.15

**Expected behavior:** 
Always increment by 0.01

**Actual behavior:**
Skips some numbers.

NickDevoctomy, 2020-11-07:
Hey there, so this may or may not be relevant to you, but I recently had to fix my CR6 and this was one of the side effects.

It's probably hard to believe, but my hot end was literally falling off.  The 2 little screws supposed to keep it in place were not.  This resulted in all sorts of weird behaviour and simply down to the fact that I couldn't see the hot end due to the shroud being in the way, I didn't know it was falling off.

Now it's been reattached and tightened, I have a consistent z offset.  I also think mine has been like that since I got it as it's made some weird rattling noise which has now stopped, and I no longer get the nozzle rubbing against the print like it was before.

ZanSchaffer, 2020-11-07:
@NickDevoctomy  quite possible, will check. But I still think it is an UI thing unless the Z offset is calculated each time I press the arrow button and isn't just a variable incrementing/ decrementing.

ZanSchaffer, 2020-11-07:
@NickDevoctomy found the screw that holds the heater block laying horizontal inside the the silicone sock. Will try it again. Thanks for pointing me in a direction.

NickDevoctomy, 2020-11-07:
Hopefully that helps with levelling, just realised I misread your issue though.  The UI not consistently changing with each press of the button, I've noticed that too.  It might be worth using a command directly to get the precise value you want in the meantime

https://marlinfw.org/docs/gcode/M851.html

I think it may be changing it to a value that's outside of the precision of the display but I could be wrong.

ZanSchaffer, 2020-11-07:
@NickDevoctomy Yeah in the UI. The problem is sometimes it works and sometimes it doesn't and I don't know how to reproduce it consistently.

NickDevoctomy, 2020-11-07:
So this is actually a bit weird, I've added some debug messages to check the code and output the value that the z offset gets changed to, and what gets sent to the UI.  The debug messages are sometimes out from what's sent to the UI.  Will let you know if I find anything more.

Sebazzz, 2020-11-07:
Well, while fixing this issue and starting a new print my mainboard died unfortunately. I can upload the firmware file I compiled, hopefully it works for you. 

[firmware-20201107-164713.zip](https://github.com/CR6Community/Marlin/files/5505001/firmware-20201107-164713.zip)

NickDevoctomy, 2020-11-07:
damn :( sorry to hear that, any idea what's caused the failure? I've already had to replace my power switch.

Sebazzz, 2020-11-07:
From what I've seen the firmware on these boards can randomly go corrupt. Sometimes reflashing helps. In my case, it didn't.

NickDevoctomy, 2020-11-07:
Have you commit the source code for the changes? I'd like to see how you fixed it, I'm currently trying to work out why it seems to randomly ignore input even though it beeps.

Sebazzz, 2020-11-07:
c3618fcc5171c7e6928f4dd433c6c9e862cd7c37 and 583a7f25186ded1c05cbd25b3d0ba1480a70f2de should be the relevant changes (which I was not able to test - otherwise I'd remove the serial echoing)

NickDevoctomy, 2020-11-07:
Oh I see, a change to screen as opposed to marlin? I'll give that a test now.

NickDevoctomy, 2020-11-07:
Ignore me, I'm being an idiot! lol

Sebazzz, 2020-11-07:
Not as much as an idiot as this idiot who thought buying a Creality printer was a good idea 🙄

NickDevoctomy, 2020-11-07:
haha! honestly the Ender 3 V2 with BL touch is what the CR6 should be. And it was cheaper.  Creality seem to be cannibalising their own products.

NickDevoctomy, 2020-11-07:
Before I install this extui branch version, should I be using an updated version of the lcd firmware too?

Sebazzz, 2020-11-07:
Yes, I assumed you were using the alpha 4 build. Sorry.

NickDevoctomy, 2020-11-07:
Okay, i'll flash that first, no probs.

NickDevoctomy, 2020-11-07:
Okay, screen updated to alpha 4, and marlin updated to your extui branch and the z offset is now changing 0.1 at a time reliably and waaaaay more responsive now too.

64Spaces, 2020-11-07:
> Well, while fixing this issue and starting a new print my mainboard died unfortunately. I can upload the firmware file I compiled, hopefully it works for you.
> 

Oh no! Sorry to hear this. It’s ironic that one of their biggest supporters has issues. On mine, I replaced the power switches and replaced the tinned wires with ferrules. Fingers crossed.

Sebazzz, 2020-12-10:
Should be fixed on latest sources.

closed 2020-12-10 by Sebazzz