CR6Community/Marlin #40: [BUG] G29 failing when printing from touch screen

NickDevoctomy, 2020-11-07:
### Bug Description

When printing g-code from the touch screen, if there is a G29 command right after the G28 command, it causes ABL to fail and for the hot end to fly off to one side and grind the motors.  Running exactly the same gcode from Octoprint works fine.  I believe it may be down to the touch screen page that reports which section it is levelling.

I now have an Ender3 V2 with a BL Touch as well as a CR6 SE, and the way that handles ABL is slightly different, it also doesn't have this fault.  When levelling from the touch screen you merely get a prompt that tells you to wait until it is complete, rather than the rather superfluous progress page you get on the CR6 SE (does this serve any purpose?).

When the G29 command is streamed from Octoprint, the ABL progress page is shown, but it does not cause any issues, this may be due to the fact that the print progress screen is never shown when printing from Octoprint.

### My Configurations

Current latest CR6Community version.

### Steps to Reproduce

1. Generate GCode for any small print using Cura that has a G28 followed by a G29 in startup.
2. Stream generated gcode from Octoprint, observe correct behavior)
3. Print generated gcoode using SD card and touch screen, notice how G29 fails. and the nozzle flies off < 0,0 and causes the motors to grind until power off.

**Expected behavior:** [What you expect to happen]

ABL command to work as expected with either Octoprint or the touch screen.

Scenarios:

1. ABL option from touch screen
2. ABL (G29) command whilst printing from touch screen (print progress page shown on touch screen)
3. ABL (G29) command streamed from Octoprint

**Actual behavior:** [What actually happens]

ABL command (G29) fails whilst in "print progress page" (scenario 2 above) causing the nozzle to fly off axis and grind the motors.

#### Additional Information

Personally I would say that the ABL progress page is entirely pointless, and highly restrictive.  Not only does it have a hard coded 4x4 matrix but provides no relevant information to the user.  If I were to change my ABL to a 5x5 matrix, the page would not function correctly. Ideal behaviour would be similar to that of the Ender 3 V2 (Smith3D 2.0.x.14 firmware) where it merely shows a dialog that says "Bed Levelling, Please wait until done." when selecting Levelling command directly from touch screen, and when G29 is processed during printing; nothing is shown.

NickDevoctomy, 2020-11-07:
Just to add, I'm currently looking at the source to see if there is an obvious fix to this.  I tried preventing the ABL callback (that updates ABL progress) when the printing PRINT_PROGRESS_RUNNING page is shown but this did not work.

NickDevoctomy, 2020-11-07:
Just to add, the issue appears to occur in this sequence,

1.  G28, Homes just fine
2.  G29, starts homing point 1, does both touches, then flies into middle of bed, then back out to < 0,0

NickDevoctomy, 2020-11-07:
Any ideas how to get the SERIAL_ECHO messages to appear when initiating commands from the printer? When initiated from Octoprint I see them fine but not from the printer.

NickDevoctomy, 2020-11-07:
I'm just testing the extui branch with alpha 4 lcd firmware to see if the problem exists there.

NickDevoctomy, 2020-11-07:
Just run a few tests and it seems that it's not starting the print from the card, it isn't heating the bed & end, just sits there with the time going up at the moment.

Sebazzz, 2020-11-07:
Do you have a log from octoprint or another host on this?

Sebazzz, 2020-11-08:
[firmware-20201108-112356-hotfix1.zip](https://github.com/CR6Community/Marlin/files/5506154/firmware-20201108-112356-hotfix1.zip)

Please try this. My mainboard is unfortunately dead, so I could not reproduce it, but this might resolve it.

NickDevoctomy, 2020-11-08:
Thanks @Sebazzz I shall give it a shot later on today.  Unfortunately with respect to logs, nothing was coming back in Octoprint, so I started adding some additional logs, but these wouldn't show when printing started, only when running G29 separately.  Is there a way to enable logging during printing?

Sebazzz, 2020-11-08:
Yes check the release notes

Met vriendelijke groet,
Sebastiaan Dammann

Sebazzz, 2020-11-08:
[firmware-20201108-125953-hotfix2.zip](https://github.com/CR6Community/Marlin/files/5506271/firmware-20201108-125953-hotfix2.zip)

If hotfix1 will not fix it, please try hotfix2 instead.

NickDevoctomy, 2020-11-08:
Okay, so first hotfix appears to be working but it's dropped the temps during bed levelling, from  200 / 65, to 120 / 50.

NickDevoctomy, 2020-11-08:
Once it finishes, it then attempts to print at those temps.  Obviously that won't work as the filament won't melt that low.

Sebazzz, 2020-11-08:
Level before you set the temperatures, for instance, using this gcode: 

https://damsteen.nl/blog/2020/10/25/creality-cr6-start-print-without-drooping

NickDevoctomy, 2020-11-08:
okay, not sure that will work with Cura, as I set the start gcode, all of the temps are put in automatically before that.  I'm not worried about drips of filament on the bed tbh, so I'll just mod my local copy of the source.

NickDevoctomy, 2020-11-08:
Thanks for your help, this is great.  You're a diamond.

closed 2020-11-08 by NickDevoctomy

Sebazzz, 2020-11-08:
> 
> 
> okay, not sure that will work with Cura, as I set the start gcode, all of the temps are put in automatically before that. I'm not worried about drips of filament on the bed tbh, so I'll just mod my local copy of the source.

When Cura doesn't detect any temperature placeholders, it will insert the temperature itself.

NickDevoctomy, 2020-11-08:
oh thanks, I wasn't aware of temperature placeholders so I shall take a look at that rather than modding my source..