CR6Community/Marlin #41: [BUG] Bed leveling fails

drbobbob, 2020-11-07:
I just flashed the Community Release 4 version on my printer.  I also updated the touch screen.  When I go to bed level, the homing sequence runs, the temperature targets get set, and once the temperature is reached it heads for position 1, pauses a bit, and then returns to the center of the bed.

I have tried reflashing the firmware using different SD cards.  I have tried running the leveling with and without an SD card.  I have tried running the leveling from a "print" file.  Nothing seems to make it want to properly level.  I do not believe I have a hardware issue because prior to reprogramming the stock firmware worked and during the homing sequence the nozzle taps the bed as expected.

I have not tried any debugging with PC connected because I don't have any USB power isolators and I don't want to risk my laptop's USB port.

Will revert to community release 3 and see if that works for me.

drbobbob, 2020-11-07:
Community Release 3 had the same issue.  Reverting to Creality firmware 1.0.3.6 allowed bed leveling to proceed

Sebazzz, 2020-11-07:
Sorry, without any information this becomes incredibly hard to debug and I must close this issue.

The commands stated in the release notes can be put in a gcode file as well and "printed". 

Also remove all attachments from your bowden tube, turn off the led and lights in your room and observe (and record) the LED in your strain gauge as the leveling process happens.

drbobbob, 2020-11-07:
I realize I was short on details, wasn't sure what would be valuable to collect.  I wrote a gcode file to attempt manual leveling, had the same failure.  I pulled it from the comments in issue 21.  Looking at other closed issues, I might be hitting what was discussed in issue 32.

I noticed that the led in the strain gauge was on much longer during the homing with the creality firmware.  I am currently printing with the creality firmware, when it is done I can record the strain gauge on the LED.  What is the best way to send the video?  youtube link?

I have the stock clips holding the bowden tube to the electrical cable.  I noticed that they have slipped to be closer to the hotend now.  I can try removing them and seeing if that affects things as well.

Sebazzz, 2020-11-07:
Please also try adjusting the little potmeter in daughter board next to the hotend.

Met vriendelijke groet,
Sebastiaan Dammann

drbobbob, 2020-11-07:
Which way should I adjust it?

It seems like you are suggesting this is an issue with faulty triggering of the strain gauge.  Removing anything that could tug on it might help.  I assume the pot on the daughter board adjusts sensitivity there.

Will be a bit before I can test these changes.

drbobbob, 2020-11-07:
I unhooked everything from the Bowden tube and same result. When homing, the blue led on the daughter board came on while traveling and when the nozzle hit the bed. Then after warming up, it moved to above position 1, the blue light flashed, and it moved back to center. Will adjust the pot and try again

drbobbob, 2020-11-08:
Was able to adjust the pot and get it to work.

For others facing this issue, it appears that the pot adjusts the sensitivity of the strain gauge. Turning it counterclockwise reduces the force needed. Turning it clockwise increases the force needed. It doesn't take much adjustment to make a change. I may dial mine in some more because right now there is some flex before the gauge registers. I had it set once too high and had to kill the printer because it seemed like it was pushing too hard. On the low side, you can get it to where the blue light is on even when the nozzle is in the air. Apparently there is a range that it will be good enough for homing but too sensitive for leveling.

drbobbob, 2020-11-08:
This is a picture of my before setting, I probably turned it less than 10 degrees clockwise to fix the issue
[image]

Sebazzz, 2020-11-08:
I got an idea how to fix it. We need to pull COM_PIN low during X/Y moves.

Met vriendelijke groet,
Sebastiaan Dammann

[20201107_175026_compress50]<https://eur02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F319549%2F98453952-762afc80-2124-11eb-94df-3a5b4a6a48b7.jpg&data=04%7C01%7C%7C252951c40bda494d535e08d8837aaf4f%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637403910226076087%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=tBd9%2BZWijSSmbZD0LZh6%2B7lnoiOkkKfhyCskN8EYfXY%3D&reserved=0>

Sebazzz, 2020-11-08:
[firmware-20201108-112356-hotfix1.zip](https://github.com/CR6Community/Marlin/files/5506155/firmware-20201108-112356-hotfix1.zip)

Please try this.

drbobbob, 2020-11-08:
I tuned the pot back to where my original symptoms occurred (homing worked, probing didn't work) with the release 4 firmware.

I flashed hotfix 1 and without changing pot settings, was able to home and probe successfully.
I flashed hotfix 2 and without changing pot settings, was able to home and probe successfully.

For reference, when I flash the new firmware, I put the bin file on the root of the SD, power cycle machine, navigate to reset to factory defaults, power cycle machine again, and test the leveling.

I'm going to keep using a hotfix firmware for now, I assume that the fix for this will come in the next release.  Do you have any recommendation for 1 vs 2?

Thanks for the quick resolution.  Sorry my initial report was vague.

Sebazzz, 2020-11-08:
Hi @drbobbob 

Hotfix2 should be fine. These are just attempts, because my motherboard is dead so I can't do anything regarding firmware development except trying something and hoping it works on someone else machine.

drbobbob, 2020-11-08:
Sorry to hear about your motherboard.  How did it die?  I am tempted to do a survey of CR6 owners to see how they have used their printers and how they have died.  I'm trying to figure out if it is random chance, failure after X hours of total runtime, failures after a single run > X hours.

Sebazzz, 2020-11-08:
>  How did it die? I am tempted to do a survey of CR6 owners to see how they have used their printers and how they have died.

I had just finished two successful prints, and was about to start another one. The printer froze completely.
I decided to power cycle it, and after that it did not want to boot anymore. 

No response over serial.

There is actually already a list of issues I've compiled here, might be useful to skim through it: https://gist.github.com/Sebazzz/ff4d716c8d2ad9bab1e87b3fc4238281

ShaneK, 2020-11-09:
Hey! I was also having this issue where I couldn't level my bed. Flashing Hotfix 2 worked for me. Thanks!

shinmai, 2020-11-10:
It might be helpful to add the instructions on tuning the strain gauge sensitivity to the release notes.
I don't know what causes the difference w/ Creality firmware, but I had to flash the hotfix firmware to get ABL to work, but afterwards the sensorboard LED is lit almost all the time, flickering off when the bowden tube flexes around slightly, and fully turning off only if applying (very) slight pressure on the bowden coupling with a finger. I for one took a minute to find this issue and figure out I'll probably need to tune the sensitivity down to stop this from happening. I'm not sure how ABL even succeeded with the gauge as senstivie as it seems.

In fact, as it is now, the release notes actually suggest turning the sensitivity _UP_, instead of down.  If I've misunderstood how the code works, and the board triggering while not running bed levelling isn't an issue, that might also be worth mentioning together with the other info on the stress gauge and ABL.

Sebazzz, 2020-11-10:
> In fact, as it is now, the release notes actually suggest turning the sensitivity _UP_, instead of dow

Correct, because the potmeter isn't a sensitivity setting but rather appears to be threshold setting. Taring the probe is done from firmware, so you should be able to set it a bit higher.

shinmai, 2020-11-11:
Ah, makes sense. So just to clarify: the LED lighting up when the nozzle isn't making contact with anything isn't an issue if ABL runs fine? Should I _still_ be able to tune it to trigger more easily or is that for people who's gauges have the trimpot set absurdly low at the factory?

Sebazzz, 2020-11-11:
In fact, with the community firmware I have been able to trim the threshold down. That allows the hotend to flex much less.

Met vriendelijke groet,
Sebastiaan Dammann

drbobbob, 2020-11-11:
I found that I had to turn the threshold up slightly (but only slightly) with the community firmware.  My current threshold setting has a slight amount of movement from the nozzle before triggering, but not much.

closed 2020-12-10 by Sebazzz