CR6Community/Marlin #42: [BUG] Octoprint receives error "kill()" from printer

Marow, 2020-11-11:
Hi, sometimes when i click Auto Home on the printer and will start than a print or Bed Visualizer with Octoprint, it kicks me out and gives error "kill()".

On the "original" firmware it was working

Also the Auto Home is very weird i think...

The Auto Home touches the bed, than goes up, than goes slow down to touch again, than goes much up and than goes down to the Z-Offset Level... 

Why it goes up so much to than goes down to 0.22mm in my case

Sebazzz, 2020-11-11:
Which firmware?

Marow, 2020-11-11:
> Which firmware?

sry..

firmware-20201109-211912-hotfix3.bin

Sebazzz, 2020-11-11:
Try adjusting the strain gauge please, like mentioned in this issue: https://github.com/CR6Community/Marlin/issues/41#issuecomment-723512018

Next time, please read the release notes thoroughly, because it has instructions on troubleshooting and how to collect logging and report issues. Help us help you 😉

closed 2020-11-11 by Sebazzz

eagleeyedpixel, 2020-11-12:
Hi team,

Just a bit of further information on this one, have worked out it will consistently kill, at least on mine if it tries to home while the optical sensor is triggered. 

To fix I just always ensure the z is high enough that the optical sensor is clear, at that point will home fine.

Latest firmware without the hot fix.

Sebazzz, 2020-11-12:
Can you reproduce it on the hotfix firmware?

eagleeyedpixel, 2020-11-12:
Installing now, will let you know.

eagleeyedpixel, 2020-11-12:
So, decided to try and replicate it again on non hot fix first. Tl.Dr: Strain Gauge likely did it. 
Results:
Turned on printer, moved z down by hand to trigger optical then homed via screen - it now worked fine.
Moved it via screen to trigger optical, then homed via Octo - also worked fine.

Moved back to trigger optical, reset printer and then tried to home, no issues again.

Tried a couple more things, but now not happening at all. 

I had already been careful of the optical before I fixed my strain gauge and been making sure it wasn't triggered before my prints so had not had the issue for a couple of days, so guessing the kill had something to do with the strain gauge being more sensitive and giving a confusing reading when optical was triggered, as I can't replicate it now and the only thing I have changed is removing the printed clips and making it slightly less sensitive.