CR6Community/Marlin #43: Firmware not installing

StuartWaudby, 2020-11-12:
### Bug Description

Firmware does not install, nothing happens.

### My Configurations

Creality CR6 SE
Currently installed firmware 1.0.3.6

### Steps to Reproduce

I follow the instructions for installing the firmware:
1. Format card
2. Copy firmware-20201025-162335.bin file onto root folder
3. Eject
4. Insert in CR6
5. Cycle power

**Expected behavior:**

Firmware to install

**Actual behavior:**

Printer boots without installing firmware.

Sebazzz, 2020-11-12:
Without pictures, logging or anything I can't do much.

closed 2020-11-12 by Sebazzz

StuartWaudby, 2020-11-12:
I could provide a photo of a printer booting up if you wish, not sure it will help.

How can I provide a log?

Sebazzz, 2020-11-12:
Please check and read the release notes.

StuartWaudby, 2020-11-12:
I have, what am I missing?

Sebazzz, 2020-11-12:
From the release notes:

## Reporting issues

Follow the guidelines below for reporting issues - report them on Github because it allows us to watch a single place for issues and workarounds might already be available:

- Open [an issue on the Github repository page](https://github.com/CR6Community/Marlin/issues) if you have found no issue matching your problem.
- Mention which release you are on, and if you compiled a release yourself (and in that case: what changes you have done).
- Be specific in reproduction steps. Include pictures or a video if useful. Ensure you have been able to reproduce it.
- Check if the issue has not been reported in Marlin, nor that you are using a slicer known to have problems (Cura 4.7 or higher).
- You can collect logging using [`M111 S247`](https://marlinfw.org/docs/gcode/M111.html) and [`M928 log.txt`](https://marlinfw.org/docs/gcode/M928.html) - please collect logging, without it is practically impossible for us to help you. Use Pronterface or Octoprint to capture serial terminal output.

StuartWaudby, 2020-11-12:
The issue was caused by the SD card. Larger SD card worked fine.