CR6Community/Marlin #44: [BUG] Homing Failed! if try to set new Z Offset immediately after Bed Leveling

Thinkersbluff, 2020-11-13:
With the Community Version 4 alpha firmware loaded (either baseline or hotfix 3) - 
If I select Auto-Level Bed and wait for it to complete, it returns to the Level menu upon completion, without waiting for me to push the "Back" button first.
If I select the Z Offset button after the Auto-Level function has completed, the printer will Auto Home, slowly lower the Z axis until the optical switch is triggered, raise the Z axis briefly, then start to descend again, but it stops before the nozzle reaches the bed and displays a bright yellow screen that says, "Homing Failed !".  I then have to cycle power to regain control of the printer.

If, instead, I select Z Offset on the Leveling menu first, after powering up, the Z Offset function works correctly.
I can also select Auto-Level Bed after doing the Z Offset and it works as described above.   

This is repeatable.

### My Configurations
[My Configs.zip](https://github.com/CR6Community/Marlin/files/5533794/My.Configs.zip)

I have flashed firmware-20201109-211912-hotfix3.bin to the motherboard.
I have flashed the v4 DWIN_SET to the display

### Steps to Reproduce

1. Power up printer
2. Select "Level" menu
3. Select "Auto Level Bed"
4. Wait for Auto Level to finish (returns to Level menu by itself)
5. Select "Set Z Offset"
6. Wait for printer to place nozzle at current Offset value above bed, and to present the adjustment buttons
7. Printer instead stops before touching bed and displays, "Homing Failed!"
8. Cycle power to reset printer.

**Expected behavior:** [What you expect to happen]

Expected printer to stay on display of 16 points grid at end of Auto-Level Bed routing, until I press "Back"
Expected printer to complete the Z Offset function normally.

**Actual behavior:** [What actually happens]

Printer automatically exits Auto-Level Bed routine after measuring 16th point. Does not wait for "Back" button to return to the "Level" menu.
When I select the Z Offset button on "Level" menu after first leveling the bed, the printer stops and displays, "Homing Failed" instead of returning to the "Level" menu.

#### Additional Information
[image]
[image]

Thinkersbluff, 2020-11-13:
[Homing Failed.zip](https://github.com/CR6Community/Marlin/files/5534477/Homing.Failed.zip)
Best I can do for a video clip showing the bug in action.  Used time compression to speed up the boring bits.  The transition from 16-point matrix to Level menu is shown in "real-time".

Thinkersbluff, 2020-11-13:
Additional Info:  I also had a Homing Failed incident this way:
1. Set Z-Offset using the Level Z Offset function
2. Perform Auto Bed Leveling
3. Using Octoprint Terminal, send M303 E0 S220 U
4. Using Octoprint Terminal, send M303 E-1 S65 U
5. Initiate printing of a model sliced in Cura 4.6.2, with the Creality Start GCode.
6. G28 fails to lower the nozzle all the way to the bed, at the bed center.  Homing Failed! screen is displayed.

ZanSchaffer, 2020-11-13:
This is probably because of the sensor on the left side of the bed being triggered when the homing starts. Check the red lights if it is on and try again.

Sebazzz, 2020-11-13:
We will know for sure when @Thinkersbluff collects a log like detailed in the release notes.

Thinkersbluff, 2020-11-13:
> We will know for sure when @Thinkersbluff collects a log like detailed in the release notes.

Oops. My bad.  What is a "log" and what do you need to see in it?  I use Octoprint.

Thinkersbluff, 2020-11-13:
> This is probably because of the sensor on the left side of the bed being triggered when the homing starts. Check the red lights if it is on and try again.

I am not clear on your thinking, here. Once I have the log uploaded, I would appreciate more info if it confirms your suspicions.

I did make one adjustment to my system that may have an impact here - I tweaked RP1, just the slightest bit (half an eyelash's width) counterclockwise.  

- Before the adjustment, the hot end would press slightly into the bed during probing, then the system would sense the bed (blue light on) and pull the nozzle back up.
- Now (after the adjustment), I do not see the hot end push the bed down at all.
- I flashed v4 alpha after making that adjustment, so I do not have before/after data  to support a theory that this adjustment is related, but I suppose it is possible.  TBD.

NOTE: I was amazed that such a slight adjustment of that screw clockwise caused the nozzle to push into the head and still not turn on the blue light, and an equally slight adjustment counterclockwise caused the light to stay on all the time, even when the head was up off the bed by 20mm.  As an Electrical Engineer, it looks to me as though that potentiometer is not sized correctly, if the circuit is basically using it as a bridge to set a detection threshold.  There should be at least a 1/4 or 1/2 turn of adjustment range in it, surely, not 1/10 or less....

Thinkersbluff, 2020-11-13:
I have tried 5 times, now, to capture a log.txt file - they just keep coming out empty, sorry.
Here I attach a report I captured from the Octoprint Terminal screen + my comments. Hope it is enough.
[Bug#44Test Report.txt](https://github.com/CR6Community/Marlin/files/5539259/Bug.44Test.Report.txt)

Sebazzz, 2020-11-13:
> I did make one adjustment to my system that may have an impact here - I tweaked RP1, just the slightest bit (half an eyelash's width) counterclockwise.

Yes, that might have caused it. Try flashing the latest hotfix from the release page if you didn't already and readjust the potmeter. As you've noted, it only needs a very slight change. If you turn it too far, the function of the strain gauge inverses and you need to pull on the hotend (instead of pushing) to let the LED turn off.

Thinkersbluff, 2020-11-13:
> Try flashing the latest hotfix from the release page if you didn't already and readjust the potmeter.

 I feel dumb, now, but I can't find any link to a hotfix release.  Is it compiled from the extui branch?  I found that.

Sebazzz, 2020-11-13:
Bottom of the page: https://github.com/CR6Community/Marlin/releases/tag/v2.0.7.1-cr6-community-release-4-alpha

Thinkersbluff, 2020-11-13:
@Sebazzz I thank you for your patience.  
I believe we can close this bug report, now.

With RP1 adjusted to require a slightly larger signal from the strain gauge when probing the bed, my printer is no longer experiencing this bug.

In my defence, the adjustment range of RP1 is _crazy_ small.  Creality really should consider increasing the range and putting out technical guidance on how to correctly calibrate it.  We are going to have all kinds of trouble when these systems get older, eccentric nuts work loose, etc..

I now have it set so that the little blue light does not come on when you touch the bowden tube (which it did) or when you home XY (which it did).  I hear a tiny tick noise again as the head first probes the bed (which I had so carefully "tuned" away) 
 
_<sigh>_

closed 2020-11-13 by Thinkersbluff