CR6Community/Marlin #45: [BUG] Screen issues

mereseydotes, 2020-11-13:
### Bug Description

Flashed firmware and screen. When I went to print from SD, there was a list of "files" that was all garbage. I was not able to select any of them. There was only one file on the card. I flashed the firmware back to the previous community release (left the screen) and was able to print, but other functions are no longer working correctly, like Z offset is always 0 and must be set every time I print and, if I select to heat the nozzle, it does not heat, I must select Preheat PLA.

### My Configurations

I'm sorry, I don't know how to do this.

### Steps to Reproduce

1. Flash firmware
2. Flash screen
3. Try to print
4. Flash to previous version of firmware
5) Try to heat nozzle, etc.

**Expected behavior:** Files show to print. Z offset is maintained. Nozzle heats when set.

**Actual behavior:** Garbage shows on screen. Z defaults to 0. Nozzle does not heat.

#### Additional Information

I'm really sorry if this is the incorrect way to do this, or if this is even the wrong place to report.

Sebazzz, 2020-11-13:
Are you a Mac user?

Unzip with the command line.

mereseydotes, 2020-11-13:
No. Windows 10. And I'm not sure what I'm supposed to unzip?

https://github.com/CR6Community/Marlin/issues/45#issuecomment-727048791

Sebazzz, 2020-11-13:
Which firmware did you exactly flash, and which screen firmware did you flash? Please link to the exact downloads.

mereseydotes, 2020-11-13:
I'm sorry, I'm new at this, though good at complaining... I followed the links in this post for the latest community firmware and also flashed the screen from what was included at the link:

https://www.reddit.com/r/CR6/comments/jpao0u/community_firmware_release_4_alpha_based_on/

When I had problems printing, I flashed just the printer, not the screen (cause it's a PITA) back to the version linked from this post:

https://www.reddit.com/r/CR6/comments/jhv8qf/community_firmware_release_3_based_on_marlin_v2071/

Thnaks!

https://github.com/CR6Community/Marlin/issues/45#issuecomment-727049563

Sebazzz, 2020-11-13:
Please check my updated reply above. Thanks. ⬆

mereseydotes, 2020-11-13:
K. Will do. The screen firmware did look like the picture included after flashing, but I believe you.

https://github.com/CR6Community/Marlin/issues/45#issuecomment-727051880

mereseydotes, 2020-11-13:
The restore factory option doesn't work. Eeek! Do you think it would work to flash the screen again and try again?

Sebazzz, 2020-11-13:
I would do the installation of community firmware 4 alpha again, carefully read the instructions, and compare the picture in the firmware ZIP file next to the README.txt.

You need these files:
- https://github.com/CR6Community/Marlin/releases/download/v2.0.7.1-cr6-community-release-4-alpha/firmware-20201109-211912-hotfix3.bin
- https://github.com/CR6Community/Marlin/releases/download/v2.0.7.1-cr6-community-release-4-alpha/CR-6-touchscreen-2020-11-06.zip

And read these instructions carefully: https://github.com/CR6Community/Marlin/releases/tag/v2.0.7.1-cr6-community-release-4-alpha
Take your time. 😉

mereseydotes, 2020-11-13:
OK, I flashed the newer firmware again, then had to turn the printer off and on again to reset factory. It just beeped, but it didn't even beep before. I couldn't read the version on the screen because it was garbage. I then flashed the screen, and I don't think this is right. I used a different card and formatted it first. Thanks!

- https://github.com/CR6Community/Marlin/releases/download/v2.0.7.1-cr6-community-release-4-alpha/firmware-20201109-211912-hotfix3.bin

- https://github.com/CR6Community/Marlin/releases/download/v2.0.7.1-cr6-community-release-4-alpha/CR-6-touchscreen-2020-11-06.zip

https://github.com/CR6Community/Marlin/issues/45#issuecomment-727059741

Sebazzz, 2020-11-13:
You're not giving a lot of information.

mereseydotes, 2020-11-13:
Did the picture not arrive? It had the message that it was uploading the files, but all the numbers after the file names were zeroes.

I'll try attaching here.
[image]

Sebazzz, 2020-11-13:
Are you can see, nothing is being flashed. Try using 7-zip to unpack the zip file. Place the DWIN_SET folder in the root of a cleanly formatted FAT32 SD card with 4096K sector size. The partition must not be too large, or the screen won't accept it.

mereseydotes, 2020-11-13:
I think it was the sector size on the card. So you're officially the best and I'm officially the worst. Now if I could just get my heatblock to stop oozing filament, no matter how many times I tighten the screws and clean it and replace the nozzle and make sure the bowden tube is inserted correctly, I could enjoy my fancy printer.

What is interesting, is that I was worried because my fans were making a loud buzzing sound when I turned the printer on and that seems to have been related to the firmware snafu.

Thank you so much!

https://github.com/CR6Community/Marlin/issues/45#issuecomment-727065695

Sebazzz, 2020-11-18:
> What is interesting, is that I was worried because my fans were making a loud buzzing sound when I turned the printer on and that seems to have been related to the firmware snafu.

Sorry, this firmware does not destroy fan bearings 😀

Since your original issue has been resolved, I will consider this issue closed. Thank you.

closed 2020-11-18 by Sebazzz

mereseydotes, 2020-11-18:
It was just wishful thinking, anyway. They must have just been deciding to be nice to me the first time I turned the printer on after flashing...

https://github.com/CR6Community/Marlin/issues/45#issuecomment-729889765

Sebazzz, 2020-11-18:
I've used your screenshot for improving the flashing instructions. Thank you!

mereseydotes, 2020-11-18:
Also, fyi - it seems like the MB doesn't require the 4096 sector size. I only have 32 GB cards (except the one that came with the printer) and that sector size is not available and the flash for the MB worked fine.

https://github.com/CR6Community/Marlin/issues/45#issuecomment-729894881