CR6Community/Marlin #46: [BUG] Bed levelling starts at 120C but screen says waiting for 200C

hej2010, 2020-11-16:
### Bug Description

Press preheat PLA and then start the bed levelling. It starts levelling directly instead of waiting for 200C, as the screen says.

### My Configurations

Using the "Community firmware release 4 alpha".

### Steps to Reproduce

1. Preheat PLA
2. Start auto bed levelling

nicholaspearson, 2020-12-06:
I've just experienced similar issues tonight for the first time with the 4 alpha.

I press print and the bed levels, then it just prints, but the bed and nozzle are not pre-heated.

Sebazzz, 2020-12-22:
> I press print and the bed levels, then it just prints, but the bed and nozzle are not pre-heated.

That seems different though - That is that G29 in your start gcode resets the temperature, doesn't it?

Sebazzz, 2020-12-23:
@hej2010 

[Creality v4.5.2 motherboard firmware-crealityv452-20201223-002158.zip](https://github.com/CR6Community/Marlin/files/5732408/firmware-crealityv452-20201223-002158.zip)
[Creality v4.5.3 motherboard firmware-crealityv453-20201223-002246.zip](https://github.com/CR6Community/Marlin/files/5732406/firmware-crealityv453-20201223-002246.zip)
[Touch screen firmware CR-6-touchscreen-2020-12-23.zip](https://github.com/CR6Community/Marlin/files/5732436/CR-6-touchscreen-2020-12-23.zip)

Can you please retry this with one of these firmware files, depending on the motherboard version you have. If you don't know or got a printer from November 2020 or earlier, you'Il most likely have the Creality v4.5.2 motherboard. 

Assuming we got it fixed I will close this issue now and consider it fixed. If it isn't fixed please provide:
- Logs captured via Pronterface/Octoprint/PuTTY with `M111 S247` entered for debug logging
- Pictures or a video as appropriate

Also, if it is fixed, I still would like to have a confirmation though😀

closed 2020-12-23 by Sebazzz