CR6Community/Marlin #47: [BUG] Canceled Prints freeze touchpad

milestyler, 2020-11-16:
[BUG] Canceled Prints freeze touchpad
When I cancel a print, the Confirm/Cancel screen comes up, but the screen freezes without being able to activate either one.

Sebazzz, 2020-11-16:
Please look in the release notes how to report a bug. For instance: I need to completely guess which release you are using.

milestyler, 2020-11-17:
I am using firmware-20201025-162335.bin.  This is my first printer and my first bug about a printer.
Steps to Reproduce:
1) Start a new print
2) Stop the print
3) Hit confirm
My touchscreen then hangs after the bed homes correctly and the touchscreen does not return to the main menu.  I am not 100% comfortable with changing GCode yet.  I just haven't worked on that skillset.  If this is specific to my printer, that is good.

Sebazzz, 2020-11-17:
Alright. This has been fixed in the alpha 4 firmware. We're not going to backport the fix, as the extui branch is completely different.

closed 2020-11-17 by Sebazzz