CR6Community/Marlin #48: [BUG] Power cycle resets Z offset to .20

mereseydotes, 2020-11-17:
### Bug Description

Since (finally correctly) flashing this version of the firmware, I've found that I've had to adjust my Z offset down, as anticipated. After each power cycle, it resets to .20. I've been adjusting it in the levelling screen. I'm not sure if it does the same if adjusted by tuning from the print screen. It may or may not be relevant that I power cycle with a surge suppressor, rather than using the printer power switch. I'm also honestly not certain whether my levelling mesh is saved, as I think my bed is pretty level overall and I've been doing mostly small prints in the center of the bed recently.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Adjust Z offset to a value lower than .20 using Levelling screen
2. Power cycle
3. Z offset is reset to .20

**Expected behavior:** Z offset should be retained

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Sebazzz, 2020-11-17:
What is "this version of the firmware", because there have been several bin files.

Please also check this: https://github.com/CR6Community/Marlin/issues/35#issuecomment-727247047

mereseydotes, 2020-11-17:
I read that one, but it didn't seem relevant. I'm able to adjust Z by .01, but it resets every power cycle. I don't think I can check the version while I'm printing. Looking at my downloads, I've got this:

CR-6-touchscreen-2020-11-06.zip
firmware-20201106-192839.bin

but, I've also got this:

firmware-20201109-211912-hotfix3.bin

I'm honestly not sure if I applied the latter. I'm going to kick myself if I've wasted your time again, though.

https://github.com/CR6Community/Marlin/issues/48#issuecomment-729164232

Thinkersbluff, 2020-11-17:
"After each power cycle, it resets to .20."

@mereseydotes - If it always resets to 0.20, that sounds like it always defaults back to the setting loaded with the firmware & never resets to a modified version.  I am currently running a 10-hour print, so I cannot go test my own printer yet, but I will edit this post with what I can learn by trying to reproduce your results.  From the file dates, it sounds like you are probably either running the v4 firmware or the hotfix3 firmware (I am sure it will help @Sebazzz if you deliberately reflash with one or the other version of the motherboard firmware and recheck for this bug.)  

If you exit from the Z-Offset adjustment screen (press "Back"), does the system display 0.20 or your new offset value?
If you then cycle power, what does the system display as your Z-Offset?

mereseydotes, 2020-11-17:
If I exit the Z offset screen, it shows my current Z offset that I just set. If I power cycle, it goes back to .20. I've been going down by .05 (really showing my appreciation for all the hard work in making the adjustment by .01 available) per print since finally getting the firmware flashed correctly. I'm currently on .00, so it's happened 4 times (and tapping the little button 20 times is what finally made me overcome my shame from my previous gaffe and open a bug report). My print should be done later today, so I can check the exact firmware version then.

https://github.com/CR6Community/Marlin/issues/48#issuecomment-729195085

Sebazzz, 2020-11-17:
[firmware-20201117-214557.zip](https://github.com/CR6Community/Marlin/files/5556131/firmware-20201117-214557.zip)

Obviously I'm still not able to test this myself, but please try this 😊

mereseydotes, 2020-11-17:
Will do as soon as I can. I saw your MB died. 

https://github.com/CR6Community/Marlin/issues/48#issuecomment-729204369

mereseydotes, 2020-11-18:
We have a winner! Thank you so much! I'm about to start my first multiday print in about 4397234907 years (at least) tomorrow AM (finally figured out that my hands are too greasy to tell if I pushed the bowden tube in all the way, so I need to start with the tube pushed in all the way, then push it up by installing the nozzle) and I'm stoked!

closed 2020-11-18 by mereseydotes

Thinkersbluff, 2020-11-18:
@Sebazzz I see this bug is now closed, though I would have thought it was rather a known issue with v4 alpha.  I was able to confirm that my system (running firmware compiled by me from the extui branch as of 2 days ago) does preserve the modified Z Offset value if one performs an AutoBed Level after entering the new Z Offset and before cycling power, which may provide a viable workaround until the next release...

I am also able to confirm that firmware-20201117-214557.zip does indeed save the new Z Offset value when power is cycled before pressing "Back", after pressing "Back" or after performing a subsequent Auto-Level bed.  
I did not regression-test any other functionality.

mereseydotes, 2020-11-18:
Sorry was I not supposed to close it? Where I work we close crs after testing them

Thinkersbluff, 2020-11-18:
> Sorry was I not supposed to close it? Where I work we close crs after testing them

@mereseydotes - I understand - My experience has been that bugs are not closed until the software version that closes them is formally released (or until we establish that the bug was reported in error.)  As far as I know, the latest release is still v4 (alpha) or v3 or hotfix3, all of which still exhibit this bug.  Anyone trying to decide which version to use may not notice this issue if the bug report is closed.  

Not my call, just sayin...

@Sebazzz - do you have any guidelines on whether/when to close bug reports, here?

mereseydotes, 2020-11-18:
I guess it's a matter or organizational preference. Where I work, we don't release software until all the bugs are closed. And now I've message up again when youall are being awesome.

https://github.com/CR6Community/Marlin/issues/48#issuecomment-729848684

yokusan, 2020-11-24:
Dear Sebastiaan, could I get a Zip with the source code for firmware-20201117-214557? I want to compile it for my E3D V6 Hotend.

Sebazzz, 2020-11-24:
You can download it from Github. Everything has been pushed.

yokusan, 2020-11-24:
Which branch is the most recent? I am not a Github Expert, sorry.

Sebazzz, 2020-11-24:
Development is done on `extui`. If you flash firmware, be sure to take the latest screen firmware as well as mentioned in the alpha 4 release notes.

yokusan, 2020-11-24:
I have the 20201106 LCD Firmware installed. Thank you for your help.

ThisIsMyUniqueUsername, 2020-12-25:
Apologies if this is a stupid question, but I've got firmware-20201106-192839.bin and I experienced the same issue with Z offset resetting to 0.2 after a power cycle.

Do I understand correctly that this has been fixed in the latest release I can compile myself, or simply wait for the official .bin file release?

Thank you.

Sebazzz, 2020-12-25:
Please follow the issue template to provide all necessary information.

Met vriendelijke groet,
Sebastiaan Dammann

ThisIsMyUniqueUsername, 2020-12-25:
> Please follow the issue template to provide all necessary information. Met vriendelijke groet, Sebastiaan Dammann
> […](#)

Thanks but I didn't want to duplicate the bug report, just asking if the bug described by mereseydoteswill be fixed in the next release.

Thinkersbluff, 2020-12-25:
Yes, this problem is fixed in the latest extui branch files and will be fixed in the next formal release.  If you are able to compile your own firmware, you can download the files from GitHub using the Code button.
Merry Christmas.