CR6Community/Marlin #51: Marlin CR-6SE Alpha-4 leaves printer prompts in chinese

oliverhbailey, 2020-11-24:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Sebazzz, 2020-11-24:
You need to flash the touch screen firmware. Our supplied touch screen firmware has no Chinese at all.

closed 2020-11-24 by Sebazzz

oliverhbailey, 2020-11-25:
Creality sent replacement board. Updated as per instructions on github. Lcd
firmware then controller firmware. Cannot read any files on any sdcards, on
firmware files. Same problem old controller had.

Not spending any more time on this printer.

Sebazzz, 2020-11-25:
The LCD firmware needs to be flashed on the LCD: https://github.com/CR6Community/CR-6-touchscreen/blob/extui/src/README.md