CR6Community/Marlin #52: [FR] Undependency from the mainboard

mrv96, 2020-11-25:
Hi, i have a Longer LK4 Pro which has the same LCD of CR6, hence i tried to use your GUI on my printer. I had some trouble to compile your code for the original atmega2560 based board, and finally it compiled well but the screen stayed on initial boot picture.

I know that yout work is about CR6, but since your code is very promising you could think to write your code undependently to the mainboard. In this way your work could be used also on another printers. It would be a shame close this code adapting it on a single printer

dryphi, 2020-11-25:
Our main goal for this Marlin fork is to clean-up the Creality source code so it can be merged upstream. As a Marlin fork, yes, we intend to merge with Marlin main. Is that what you're suggesting?

dryphi, 2020-11-25:
Also this probably belongs in the touchscreen fork:
https://github.com/CR6Community/CR-6-touchscreen

mrv96, 2020-11-25:
No, this issue is not about touchscreen firmware.

I mean that it would be preferable if your Marlin fork works compiles and works with any mainboard. Try to compile it for an atmega2560 based board (e.g. ramps 14) and you'll see why I opened this issue

https://github.com/CR6Community/CR-6-touchscreen

Sebazzz, 2020-11-25:
Why would you want to use this fork instead of mainstream Marlin? 

This fork is meant for:

- Providing up to date and stable Marlin for the CR-6 SE native and [BTT SKR CR6](https://damsteen.nl/blog/2020/11/25/how-to-btt-skr-cr6-installation) motherboard
- Expanding the features of the limited touch screen

Once upstream Marlin supports the strain gauge, [currently being whipped into shape in this PR I've submitted](https://github.com/MarlinFirmware/Marlin/pull/19958), the future of this project will probably be:

- Still expanding the features of the touch screen
- Provide builds for the CR-6 and SKR boards for the less technically inclined

mrv96, 2020-11-25:
I wrote here because:
- mainstream Marlin has a very limited support to DWIN display
- your code is good and it would be a shame to keep it limited to a single printer only
- you made a PR to official Marlin and it is not printer specific, but it is general for each printer. Hence, IMHO if you want to merge with mainstream, it would be better to generalize and open your code to every printer. So far this was the Marlin philosophy I guess

Sebazzz, 2020-11-25:
> - mainstream Marlin has a very limited support to DWIN display

I actually based the implementation on [that of upstream Marlin](https://github.com/MarlinFirmware/Marlin/tree/2.0.x/Marlin/src/lcd/extui/lib/dgus) but it was too opinionated to use for how the CR-6 display was constructed so I still had to change most of it. The original code was implemented in [this PR](https://github.com/MarlinFirmware/Marlin/pull/13253).

> you made a PR to official Marlin and it is not printer specific, but it is general for each printer. Hence, IMHO if you want to merge with mainstream, it would be better to generalize and open your code to every printer.

I don't understand this?

mrv96, 2020-11-25:
I mean: if you can, please, generalize your code allowing it to be executed also on non-CR6 printers because it is very promising. This one https://github.com/MarlinFirmware/Marlin/tree/2.0.x/Marlin/src/lcd/extui/lib/dgus is worst than yours.

Otherwise you can close the issue.

Thank you.

Sebazzz, 2020-11-25:
> I mean: if you can, please, generalize your code allowing it to be executed also on non-CR6 printers because it is very promising. This one https://github.com/MarlinFirmware/Marlin/tree/2.0.x/Marlin/src/lcd/extui/lib/dgus is worst than yours.

I don't see how that _can_ be worse because the code in this repo is _based_  on that one. I'm happy to help and think along, but what you're asking is very vague and not very actionable 😉

mrv96, 2020-11-25:
It is worst because have too few features. For instance, it doesn’t provide on-screen ABL. Yours yes. That was only an example refered to one specified feature, but of course it can be applied to the others.

Sebazzz, 2020-11-25:
Okay, but you should be able to compile this for any other printer just fine by changing the default_envs in platformio.ini and the correct changes in Configuration.h.

mrv96, 2020-11-25:
As you can seen, it is not true:

```
> Executing task: C:\Users\mario\.platformio\penv\Scripts\pio.exe run <

Processing mega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR (3.0.0) > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES:
 - framework-arduino-avr 5.1.0
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...    
Dependency Graph
|-- <Wire> 1.0  
|-- <SPI> 1.0   
|-- <EEPROM> 2.0
|-- <SoftwareSerial> 1.0
Building in release mode
Compiling .pio\build\mega2560\src\src\HAL\AVR\HAL.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\HAL_SPI.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\MarlinSerial.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\Servo.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\eeprom.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\fast_pwm.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\fastio.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\u8g_com_HAL_AVR_sw_spi.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\AVR\watchdog.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\HAL_spi_L6470.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\eeprom_api.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\eeprom_if_i2c.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\eeprom_if_spi.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\esp_wifi.cpp.o
Compiling .pio\build\mega2560\src\src\HAL\shared\servo.cpp.o
Compiling .pio\build\mega2560\src\src\MarlinCore.cpp.o
Compiling .pio\build\mega2560\src\src\core\serial.cpp.o
Compiling .pio\build\mega2560\src\src\core\utility.cpp.o
Compiling .pio\build\mega2560\src\src\feature\babystep.cpp.o
Compiling .pio\build\mega2560\src\src\feature\bedlevel\abl\abl.cpp.o
Compiling .pio\build\mega2560\src\src\feature\bedlevel\bedlevel.cpp.o
Compiling .pio\build\mega2560\src\src\feature\caselight.cpp.o
Compiling .pio\build\mega2560\src\src\feature\e_parser.cpp.o
Compiling .pio\build\mega2560\src\src\feature\host_actions.cpp.o
Compiling .pio\build\mega2560\src\src\feature\pause.cpp.o
Compiling .pio\build\mega2560\src\src\feature\powerloss.cpp.o
Compiling .pio\build\mega2560\src\src\feature\runout.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\bedlevel\G26.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\bedlevel\G42.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\bedlevel\M420.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\bedlevel\abl\G29.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\bedlevel\abl\M421.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\calibrate\G28.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\calibrate\M48.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M200-M205.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M220.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M221.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M301.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M302.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M304.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M43.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\config\M92.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M108_M112_M410.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M111.cpp.o
Compiling .pio\build\mega2560\src\src\gcode\control\M120_M121.cpp.o
[... 189 lines cut ...]
   #define DEBUG_ECHOLNPAIR        SERIAL_ECHOLNPAIR
                                   ^
Marlin\src\module\../inc/../core/macros.h:89:27: note: in expansion of macro '_BV'
 #define TEST(n,b) (!!((n)&_BV(b)))
                           ^
Marlin\src\module\../inc/../HAL/./AVR/fastio.h:69:31: note: in expansion of macro 'TEST'
 #define _READ(IO)             TEST(DIO ## IO ## _RPORT, DIO ## IO ## _PIN)
                               ^
Marlin\src\module\../inc/../HAL/./AVR/fastio.h:101:31: note: in expansion of macro '_READ'
 #define READ(IO)              _READ(IO)
                               ^
Marlin\src\module\motion.cpp:1585:45: note: in expansion of macro 'READ'
       DEBUG_ECHOLNPAIR(">>> is_homing_z (", READ(OPTO_SWITCH_PIN), ")");
                                             ^
At global scope:
cc1plus.exe: warning: unrecognized command line option '-Wno-register'
*** [.pio\build\mega2560\src\src\module\motion.cpp.o] Error 1
================================================================================= [FAILED] Took 14.92 seconds =================================================================================
Environment    Status    Duration
-------------  --------  ------------
mega2560       FAILED    00:00:14.918
============================================================================ 1 failed, 0 succeeded in 00:00:14.918 ============================================================================The terminal process "C:\Users\mario\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

Sebazzz, 2020-11-25:
Please check #30 for this. Don't build with Arduino - use VSCode + PlatformIO.

mrv96, 2020-11-25:
This is VSCode + PlatformIO:
[image]

grobux, 2020-11-25:
I'm not an expert in C, but one of the first errors you get is:

> error: 'u_int8_t' was not declared in this scope

Doesn't that mean there is an include missing ? Try adding `#include <stdint.h>`, clean and build.
If `DGUSDisplay.cpp` doesn't compile, maybe it explains why other files can't compile.

Sebazzz, 2020-11-25:
My apologies, I don't have time to focus on this right now @mrv96. If you do get this working, please submit a pull request to the `extui` branch and I will merge it.

mrv96, 2020-11-25:
Done in https://github.com/CR6Community/Marlin/pull/53. Now the code compile also for atmega2560 boards.

Sebazzz, 2020-11-25:
Merged. Thank you! 

So, you are using the Creality stock TFT on a different printer?

closed 2020-11-25 by Sebazzz

mrv96, 2020-11-25:
I would like to try. I tried your repo on a Longer LK4 Pro about two weeks ago and it didn't compile because a problem with serial communication. You used an STM plaform specific command that miss in AVR. I fixed it but the screen was locked on boot screen. Today I hasn't been able to get that error to fix it. I suppose that you have already changed that code.

Now that the firmware compile, I'm going to pysically test it on my printer.

Longer LK4 Pro (aka Alfawise U30 Pro) uses the same display of CR6

Sebazzz, 2020-11-25:
There is a flag in `DGUSDisplayDef.h` you can use to debug serial communication. You can see the handshakes etc. There are also a few `DEBUG_xyz` macro calls you can uncomment for further debugging. Also in the `doc/vendor` folder of the touch screen repository you can find all kinds of documentation. Good luck!