CR6Community/Marlin #54: [FR] Stepper ON/OFF behaviour should be same as for Ender-3 with LCD Panel

Thinkersbluff, 2020-11-26:
I was planning to file a bug report, saying that Stepper Control seems to be implemented as "all or none", rather than each motor enable/disable being controlled individually as it is on my Ender-3, but I suspect the CR-6 behaviour is actually coded this way by Creality for some reason.  I have therefore classed this as a Feature Request instead, for consideration.

On my Ender:
- when I power-up, all 3 stepper motors are disabled by default.
- if I move just one axis (any of X, Y or Z), only that stepper is locked at the end of the movement. The other steppers remain disabled unless and until I also command those to move.
- when I select "Disable Steppers" (which has no ON/OFF status indicator in the menu), ALL steppers are released.

On my CR-6, running the latest extui version (downloaded 10pm on 25 Nov 2020):
- when I power-up, all 3 stepper motors are disabled by default.
- when I select "Prepare" - depending upon whether Octoprint is connected or not, before I do that (see [https://github.com/CR6Community/CR-6-touchscreen/issues/18](url) - the "Disable Stepper (ON/OFF)" button is EITHER ON or OFF
- IF it was OFF, then selecting Disable Stepper causes the button to change to ON.  After that, I can find no way to make it go back to OFF, even if I move the motors, etc., but that might be because I leave Octoprint connected to enable logging.
- when I move ANY one axis, ALL 3 motors lock.
- when I select Disable Stepper ON/OFF ALL 3 motors are released.

I only own the two printers, so I have no idea whether the Ender-3 behaves the way all other 3D printers behave, or whether they are both unique, I only know they are not the same.  

**In my mind, it would be wonderful if users could have all printers running Marlin behave in one standard way.

If possible, please at least rename the Disable Stepper ON/OFF control to read "Disable Steppers", since it actually does release them all, by design, and the ON/OFF suggests a toggling function but pressing the button repeatedly does NOT toggle the state of the motors.**

NOTE: If in fact that button SHOULD toggle the state of the steppers, please note that it presently does NOT.

[Disable Stepper Bug Log.txt](https://github.com/CR6Community/Marlin/files/5604638/Disable.Stepper.Bug.Log.txt)

Sebazzz, 2020-11-26:
Thank you for the report. To summarize: the issue is that when moving a stepper motor, all steppers are locked, instead of just the stepper that has been moved?

Thinkersbluff, 2020-11-26:
Yes.

I like to Home X, and then pull the bed forward, if I have cancelled a print due to some kind of error.  Locking all 3 steppers when I move X to the left means I also have to select Back and Disable Steppers as part of my workflow...

Sebazzz, 2020-11-26:
I forgot to ask, is this when using the touchscreen or when using Octoprint?

Thinkersbluff, 2020-11-26:
> I forgot to ask, is this when using the touchscreen or when using Octoprint?

I did not test moving motors from Octoprint, to see whether it locked all 3 motors.  I will do that soon and update this report.

Is it ok/correct of me to be renaming this issue for greater precision, or should I stop doing that?

Sebazzz, 2020-11-26:
> Is it ok/correct of me to be renaming this issue for greater precision, or should I stop doing that?

It helps keep things organized, thank you 😀

Sebazzz, 2020-11-26:
Issue [CR-6-Touchscreen#18](https://github.com/CR6Community/CR-6-touchscreen/issues/18) will be "resolved" by removing the toggle, as you have already suggested.

Sebazzz, 2020-11-26:
In the touch screen we do this:

https://github.com/CR6Community/Marlin/blob/c16e2183af125d68958e6d6bcdf4a8c5de830a8f/Marlin/src/lcd/extui/lib/dgus_creality/DGUSScreenHandler.cpp#L818-L820

And a G0 indirectly does this:

https://github.com/CR6Community/Marlin/blob/c16e2183af125d68958e6d6bcdf4a8c5de830a8f/Marlin/src/gcode/motion/G0_G1.cpp#L101

which then does

https://github.com/CR6Community/Marlin/blob/c16e2183af125d68958e6d6bcdf4a8c5de830a8f/Marlin/src/module/motion.cpp#L926

So, I suppose we could just fill `destination` and call ``prepare_line_to_destination`. 

The only problem I see is that we also feed the current position back to the screen, so things might not work well and we'd need to maintain our own position buffer. Perhaps @Doridian also has some insights into this.

Doridian, 2020-11-26:
@Sebazzz 
The thing, though, is that we don't re-sync all positions.
We only sync positions in the change handler.
So the display would need to fire a change event on all 3 axes for it to update all 3 positions.
`current_position` is a Marlin internal variable, which isn't even defined by the screen.
Maybe we should add a serial log on this event to see what events fire.

Sebazzz, 2020-11-26:
I think so. It is too late to properly analyze Marlin's planner algorithms 😅 but from what I can see the steppers only get enabled when there are actually steps to perform, so we send something in the planner what probably warrants a tiny microstep.

Thinkersbluff, 2020-11-27:
> I forgot to ask, is this when using the touchscreen or when using Octoprint?

I have now confirmed that all 3 axes on the CR-6 also lock if I nudge just one of the three axes with the move buttons.

Thinkersbluff, 2020-11-27:
I see the discussion above & though it important to add these things I learned today:
1. All 3 motors on the CR-6 unlock themselves 2 minutes after I have moved the last one.
2. I discovered that tapping on the position actually pops up a keypad - did not know that before - moving them that way also behaves as described above. 
3. Neither of the other position displays change, when I move just one axis.
4. The Ender-3 is running an earlier version of Marlin - 2.0.5.3.

Thinkersbluff, 2020-11-27:
I forgot that Configuration_adv.h also allowed me to set the automatic stepper release command to M84 XYE so that my Z axis would not drop.  I make a habit of setting that on my Ender-3 and will start doing that on the CR-6 also, in case it impacts my Z-Offset calibration activity.   Does that make sense?

Sebazzz, 2020-12-10:
> It may be a kindness to make those settings a default, for the Community Version bin file users.
> Should I open a new FR to suggest that?

@Thinkersbluff Yes, please do. If there isn't anything actionable here, please close this issue.

closed 2020-12-10 by Sebazzz