CR6Community/Marlin #55: [FR] Support progress bard and file name when printing from host

ZanSchaffer, 2020-11-27:
### Description

Support progress bar and filename when printing from host
### Feature Workflow

1. Connect host
2. Change the filename and progress % with a gcode command from host

#### Additional Information

If no such command exist maybe parse M117 command from DisplaylayerProgress OctoprintPlugin with the following format
[progress]% L=[current_layer]/[total_layers]

Sebazzz, 2020-11-27:
This is supported via gcode M73 (issue #7 and #27) for progress in [release 4 alpha](https://github.com/CR6Community/Marlin/releases/tag/v2.0.7.1-cr6-community-release-4-alpha). 

File name is no gcode for, I think we should have a separate screen when printing from the host @grobux - without Resume/Stop buttons and only a button for going to the tune screen.

ZanSchaffer, 2020-11-27:
@Sebazzz Is there a standard way/plugin that can read hosts progress value and send it via M73?

Sebazzz, 2020-11-27:
@Nushio would know that better than me.

ZanSchaffer, 2020-11-27:
I found this it supports print time genius https://plugins.octoprint.org/plugins/m73progress/

ZanSchaffer, 2020-11-30:
@Sebazzz got it working you can close the issue. Unless you want to keep it open for the separate print screen when printing from host.

Sebazzz, 2020-12-10:
Not going for a separate screen now. 👍

closed 2020-12-10 by Sebazzz