CR6Community/Marlin #56: [BUG] ETA override and Tune button

giorgiomanieri, 2020-11-27:
### Bug Description

No Tune button and his function during printing.

### My Configurations

Community firmware release 4 alpha

### Steps to Reproduce

1. Send a print from Octoprint
2. Start purge extrution 
3. Start printing model

**Expected behavior:** [What you expect to happen]

Showing printing progress with ETA and Tune button on the screen.

**Actual behavior:** [What actually happens]

Time stuck to starting model time and no Tune function.
Shown on pic.

* Provide pictures or links to videos that clearly demonstrate the issue.
[image]

* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Sebazzz, 2020-11-27:
Are you injecting M73 and M75? Please check #7 and #27.

giorgiomanieri, 2020-11-27:
M75 and M77 at start and end gcode. Where M73?

Sebazzz, 2020-11-27:
M73 displays the progress.

giorgiomanieri, 2020-11-27:
> M73 displays the progress.

The progress bar works, nothing else

Sebazzz, 2020-12-10:
It sounds to me you may be injecting "print timer stop" before "print timer start".

Sebazzz, 2020-12-22:
[Creality v4.5.2 motherboard firmware-crealityv452-20201223-002158.zip](https://github.com/CR6Community/Marlin/files/5732408/firmware-crealityv452-20201223-002158.zip)
[Creality v4.5.3 motherboard firmware-crealityv453-20201223-002246.zip](https://github.com/CR6Community/Marlin/files/5732406/firmware-crealityv453-20201223-002246.zip)
[Touch screen firmware CR-6-touchscreen-2020-12-23.zip](https://github.com/CR6Community/Marlin/files/5732436/CR-6-touchscreen-2020-12-23.zip)

Can you please retry this with one of these firmware files, depending on the motherboard version you have. If you don't know or got a printer from November 2020 or earlier, you'Il most likely have the Creality v4.5.2 motherboard. 

Assuming we got it fixed I will close this issue now and consider it fixed. If it isn't fixed please provide:
- Logs captured via Pronterface/Octoprint/PuTTY with `M111 S247` entered for debug logging
- Pictures or a video as appropriate

Also, if it is fixed, I still would like to have a confirmation though😀

closed 2020-12-22 by Sebazzz