CR6Community/Marlin #57: [FR] configure e-steps directly from printer

ratwix, 2020-11-27:
### Description

Add an option to configure automaticaly e-steps in the printer without Octoprint or a USB cable plugged to the PC.

### Feature Workflow

1. Select in the menu somewhere "configure e-steps"
2. Follow the instruction (like [here](https://teachingtechyt.github.io/calibration.html#esteps)) : mark your finament @120mm, run the procedure, measure agains the mark, enter the value
3. Printer compute & save e-steps

#### Additional Information

* CR6-SE USB port is crap, if we can avoid using it, it will save a lot of USB ports & motherboards

RandyRVA, 2020-11-28:
Yes, this would be a great feature.  I killed my computer and printer by connecting the usb cable to try to do this.

dryphi, 2020-11-28:
I agree this could be helpful esp given the USB problems on the CR6.
In the meantime, this can be done from an SD card by adding commands to the start G-code.
First mark the material at a known distance from the extruder (usually 110 or 120mm). Then put the following commands into the start G-code of an otherwise blank print:
G1 E100 F100
--> Determine how much actually extrudes (mark the material beforehand)
--> Enew = Eold*100/(length actually extruded). Default E-step value from the factory is 93 I believe. So this formula becomes Enew = 9300 / (Length extruded).

With a new "blank" print, add the following commands to the start G-code instead:
M92 E###.# (enter the new value)
M500

RandyRVA, 2020-11-28:
Yes but there is no way to know what the current esteps value is without a terminal window...  I think?

dryphi, 2020-11-28:
> Yes but there is no way to know what the current esteps value is without a terminal window... I think?
As far as I know that's correct. The M503 command outputs to the terminal but not to the TFT screen. Hence my suggestion is to base it on the factory value which is 93.
Also you can adjust the extrusion multiplier (flow rate in Cura) to get close. This does essentially the same thing - just multiplies the E-steps by whatever percentage you input. Then when you're happy with the value store it using the M92 command.

Sebazzz, 2020-11-28:
This is a relatively low priority feature imho though. You only do it once, then leave it.

(Don't forget that changing e-steps is often hiding your problem of incorrect tension on the feeder. The gear ratio is fixed, the number of steps the stepper has to make for a turn too.)

ratwix, 2020-12-01:
> I agree this could be helpful esp given the USB problems on the CR6.
> In the meantime, this can be done from an SD card by adding commands to the start G-code.
> First mark the material at a known [...]

You need to know your "old" esteps, and for that, you need to plus the USB cable. So no, you can not be 100% independent with SD CARD

Sebazzz, 2020-12-01:
The default e-steps is 93 - besides you can set it to a known value and calibrate from there.

Allan-N, 2020-12-01:
Question.  Is the default e-steps value (93) one that just happens to work well for most people or was it computed based on the stepper motor and gear ratio?

dryphi, 2020-12-01:
> Question. Is the default e-steps value (93) one that just happens to work well for most people or was it computed based on the stepper motor and gear ratio?

That's the Creality default. I would assume it's based on the extruder gear diameter.

dryphi, 2020-12-01:
> You need to know your "old" esteps, and for that, you need to plus the USB cable. So no, you can not be 100% independent with SD CARD

Yes you can. Just set it to a known value such (e.g. 93) using M92 E93 then go from there.

Sebazzz, 2020-12-01:
> 
> 
> Question. Is the default e-steps value (93) one that just happens to work well for most people or was it computed based on the stepper motor and gear ratio?

Yes. You shouldn't need to change the esteps with the default extruder. Instead, tension the tensioning screw.

nicholaspearson, 2020-12-11:
> > Question. Is the default e-steps value (93) one that just happens to work well for most people or was it computed based on the stepper motor and gear ratio?
> 
> Yes. You shouldn't need to change the esteps with the default extruder. Instead, tension the tensioning screw.

I disagree, the E-Steps on almost every machine I've ever touched was different than 93.

This should be configurable from the touch interface, the CR-20 Pro 3DPC firmware allows for this and it's a life saver. So much time saved.

Default it to the current E-Steps and then allow the user to change it from there, this is 3D Printing core functionality.

dryphi, 2020-12-11:
It's a good idea. Post it to Marlin Main.

Thinkersbluff, 2020-12-12:
I already have the ability to enter parameters like esteps and driver current directly through the LCD menus on my Ender 3, so Marlin may support the functionality, but only on an LCD controller?

Sebazzz, 2020-12-12:
It is supported but not *through* the LCD. Only though gcode or host programming.

Met vriendelijke groet,
Sebastiaan Dammann

dryphi, 2020-12-17:
Suggest we merge this with https://github.com/CR6Community/Marlin/issues/60 and https://github.com/CR6Community/Marlin/issues/66

Dreambeemer, 2021-01-02:
It would be realy nice when you could change the e-steps from the screen without using the USB-port ore changing the G-codes.
I'am a totally noob in 3D-printing, so for me it should be a big releave. I'am very afraid to mess around with the G-codes, because I don't know how it works. Release 5 works like a charm. Credits for you all who worked it out.

Sebazzz, 2021-01-03:
Current working concept - ignore the user interface work for now, that's not my specialty, this is just getting the "back-end" ready:

**Heat up to target temperature**
[image]

**Extruding**
[image]

**Cue to measure remaining filament**
[image]

**New e-steps are calculated for you**
[image]

**And apply them if desired**
[image]

You can also directly enter a value in the "your new e-steps" field and apply if you don't like being guided.

Dreambeemer, 2021-01-03:
Waauw, that’s quick. I can’t wait to use it.
Great job👍🏻 Where should we be without you guys.

ratwix, 2021-01-06:
excellent ! thanks

JamesSF69, 2021-01-09:
Yes, please!!! Thank you for getting this on the agenda! Hope to use it soon!

Sebazzz, 2021-01-09:
In [#prelease-testing channel](https://discord.com/channels/759603746270609428/791082544899555358/797579692747784232) of our Discord an early build is available if anyone wants to try ([invite link if you haven't joined the Discord earlier](https://discord.gg/RKrxYy3Q9N)).

Thinkersbluff, 2021-01-10:
I tried the new esteps function with the “free air” method described in Luke’s hot end fix document. I have his fix installed. I use Capricorn tubing, an all-metal extruder cold end with the stock extruder gear and an otherwise stock CR-6 SE hot end and 0.4mm stock nozzle.
I removed the Bowden tube from the cold end of the extruder and measured the filament extruded from or retracted into the cold end outlet.

Marlin estimates the esteps for this extruder cold end to be 93, on the Ender3, from whence I took it.  It works with precisely the same gear as fitted in the stock C-6 SE extruder.

The extrude test calculated 98.7 steps if I extruded into the Bowden tube, but 95.1 if I extruded into free air. I think some advice may be necessary on how this test is meant to be performed, to help Users avoid inadvertently creating themselves a problem calibrating their extruders with friction in the system.

I also tried with and without the filament runout sensor in series with the extruder, just to be sure it did not impact the extruder’s performance. I saw no impact on the esteps..

Overall, I liked the interface. If possible, it would help to give feedback like “E-Heating” while the nozzle is heating. I was confused at first, when it said “Extruding” but it wasn’t.
I don’t like s/mm as units, since s already means seconds.  I recommend either spell it out in full or put the units in the label and just write the value.

I did notice that the esteps function extrude rate is much slower than is programmed for the load/unload function. I thought it much too slow. Is there a design reason for that rate?

Thinkersbluff, 2021-01-10:
NOTE: We "Brits" have a saying, "Much wants more."

While I was enjoying this new powerful support feature for eSteps calibration, I thought, "Hey! This would be really easy to adapt into a %Flow calibration tool!"

Sebazzz, 2021-01-26:
> The extrude test calculated 98.7 steps if I extruded into the Bowden tube, but 95.1 if I extruded into free air. I think some advice may be necessary on how this test is meant to be performed, to [...]

I'm not sure where we have a place for such guidance or if that is even up (in scope) for the firmware to do. Unless DWIN has something magical to easily adding such guidance, this would require adding pages.  That said, opinions differ on whether to use a free air method or not. These screens are obviously tailored for the latter (but also usable for the former).

> I don’t like s/mm as units, since s already means seconds. I recommend either spell it out in full or put the units in the label and just write the value.

I've removed the units completely (note: took the screenshots before I did that).

> I did notice that the esteps function extrude rate is much slower than is programmed for the load/unload function. I thought it much too slow. Is there a design reason for that rate?

I'm using the recommended speed from the Teaching Tech website.

> While I was enjoying this new powerful support feature for eSteps calibration, I thought, "Hey! This would be really easy to adapt into a %Flow calibration tool!"

You're keeping us busy 😉 I'm not sure if we want to implement a full calibration suite here. You'd still need to slice a cube in vase mode to do the actual test, and there is also not anything to save at the Marlin side.

----

Okay, current state of things:
- @grobux cleaned up my pink comic sans into something nice
- I did some changes on the Marlin side for the changes at the touch screen side

This is how it now looks:

[image]

[image]

[image]

With this I consider the feature implemented. Keep watching the #prerelease-testing channel on the Discord for a build if you want to play with it.

closed 2021-01-26 by Sebazzz

Thinkersbluff, 2021-01-26:
😚