CR6Community/Marlin #58: [BUG] Filament change M600 can't recover from 'heater timeout'

tangcla, 2020-11-27:
The heater change at layer will not reheat the extruder after it has timed out, and the button is blank.

I've attempted this twice. Unfortuantely both times I was not around to do the filament change in time for it to not timeout.
[image]

Thinkersbluff, 2020-11-28:
I believe M600 is only supported if you have an LCD screen, at present.  That is why the filament runout center only does a stop SD print right now (M25) and not an M600.

tangcla, 2020-11-28:
> I believe M600 is only supported if you have an LCD screen, at present. That is why the filament runout sensor only does a stop SD print right now (M25) and not an M600.

What do you mean by LCD screen (doesn't the CR6SE have one, which is what I took the photo of)?

Thinkersbluff, 2020-11-28:
No, what the CR-6 has is called a “touch-screen”, in Marlin.  The Ender-3 has an LCD screen, with a controller knob and additional menus.

tangcla, 2020-11-28:
Does this mean the M600 (also pause at layer height) can't resume?

Thinkersbluff, 2020-11-28:
> Does this mean the M600 (also pause at layer height) can't resume?

According to this issue record from Marlin, it may be possible to recover from M600 by sending M108:
https://github.com/MarlinFirmware/Marlin/issues/12494

According to these notes on the M108 Break and Continue command:
“With both EMERGENCY_PARSER and HOST_KEEPALIVE_FEATURE enabled, hosts will be able to prompt for continuation or cancellation, confirming with M108 and cancelling with M112.“

The problem seems to be that the M600 command causes the motherboard to ignore commands from the host, while waiting for a response from the controller.  There is a feature called Emergency Parser which should force the motherboard to process a very short list of commands while in that busy state and M108 is one of those.  

At least two other problems complicate this issue:
1. Emergency Parser is not implemented for all motherboards.  
2. If the message buffer is full, the M108 command may still be ignored.  (That’s why Thinkyhead told that 2018 bug reporter to try M108 several times)

I will test this when my current print job ends and update this response.

Sebazzz, 2020-11-28:
> As to your original BUG, I suggest that the inability of the printer to recover from a heater timeout in this instance may not be a bug so much as a fact.

I believe it is in this case though. The code uses a dialog system for the messages that are passed through the ExtUI API. I suspect something is not entirely right here. But... because in this case heating is involved I think the way it is displayed to the end user needs to be rethought.

I think I'm going to re-implement a bit that will instead:
- Navigate to the print paused screen
- Show the message from Marlin in the M117 line
- The "resume" function would then allow to execute the callback we get from Marlin

tangcla, 2020-11-28:
I'm not a software guy, but is the blank button on the screen related to the issue? Perhaps it's not linked to a command that the printer is expecting? (or it could just be a lcd firmware thing)

This also happens with the pause on layer command which I tested too. 
[image]

Sebazzz, 2020-11-28:
In my current set-up I'm unable to test this myself. Can you please try the file below? 

If I did it right you should have navigation to the print paused screen. From there you can tap "resume" to follow the filament change procedure. There should also be a temporary indication in the about screen that this file was compiled for this bug.

Please capture any detail you can if it still does not work.

[firmware-20201128-232436.zip](https://github.com/CR6Community/Marlin/files/5611314/firmware-20201128-232436.zip)

Thinkersbluff, 2020-11-29:
> Please capture any detail you can if it still does not work.

That was scary - I ended up having to reflash to Creality's v2.0.1.2, to recover the use of my printer. :~
Not sure how, but the Feed/Return got "messed-up", such that Feed 100mm gave me ~10mm instead and Return 100mm gave me an initial retraction then a feed of ~20mm more.
I tried flashing back to alpha v4 but that did not correct the problem. Nor did reflashing the display firmware.
I will take a long pause to get my heart rate back down & try going back to alpha v4 tomorrow night. 
Sorry.

Sebazzz, 2020-11-29:
> Not sure how, but the Feed/Return got "messed-up", such that Feed 100mm gave me ~10mm instead and Return 100mm gave me an initial retraction then a feed of ~20mm more.

That is an unrelated issue @Nushio is working on (the feed screen seems to be a bit off).

Sebazzz, 2020-11-29:
Also @Github, "should fix" is not the same as "fixes". Stop closing my issues 😄

Sebazzz, 2020-11-29:
[firmware-20201129-163259.zip](https://github.com/CR6Community/Marlin/files/5612738/firmware-20201129-163259.zip)

This contains some fixes in the feed menu and filament changes (thanks to the testers in the Discord).

Thinkersbluff, 2020-11-30:
> firmware-20201129-163259.zip

Help! I'm trapped in a "Tune" menu screen, while Octoprint has its way with my printer...

I like having Z offset in the Tune menu, though the box around the current value invites me to tap it to enter a value by keypad, except that nothing happens when I do.
The LED button beeps when I press it, but it still says ON and my LED stays ON.
Back beeps but I can't get back.

I wonder what layer is printing now?...

Sebazzz, 2020-11-30:
[firmware-20201201-000916.zip](https://github.com/CR6Community/Marlin/files/5619260/firmware-20201201-000916.zip)

Next attempt

tangcla, 2020-12-01:
> [firmware-20201201-000916.zip](https://github.com/CR6Community/Marlin/files/5619260/firmware-20201201-000916.zip)
> 
> Next attempt
[image]

tangcla, 2020-12-01:
> [firmware-20201201-000916.zip](https://github.com/CR6Community/Marlin/files/5619260/firmware-20201201-000916.zip)
> 
> Next attempt

Goes to pause screen, shows holding temperature (200/60) but when I click resume, it shows "heater timeout" and temperature is actually 25/0.
[image]
Also, this firmware version seems to have reset all my settings.

Sebazzz, 2020-12-01:
Did you try clicking resume again?

tangcla, 2020-12-01:
Yes, I did.
I left it on purpose so it will drop heat. The screen shows 200/60 degrees the whole time. I pressed resume, then the yellow text showed timeout error, and then it showed the 'correct' temperature.
Pressing resume does nothing (it does not change).

tangcla, 2020-12-03:
I tested the firmware-20201201-000916.zip version yesterday with a filament change layer. 
As soon as the nozzle was lifted (and the beeping completed) I pressed resume. The system rebooted and I was presented with the power loss, system start up (Creality community firmware splash screen) and then the resume print function. 
I tested it twice (same gcode file) with the same result.

Sebazzz, 2020-12-04:
Unless someone else from @CR6Community/developers has time to look at this, I'Il receive a new CR-6 soon which I should be able to use to further troubleshoot this.

Sebazzz, 2020-12-22:
[Creality v4.5.2 motherboard firmware-crealityv452-20201223-002158.zip](https://github.com/CR6Community/Marlin/files/5732408/firmware-crealityv452-20201223-002158.zip)
[Creality v4.5.3 motherboard firmware-crealityv453-20201223-002246.zip](https://github.com/CR6Community/Marlin/files/5732406/firmware-crealityv453-20201223-002246.zip)
[Touch screen firmware CR-6-touchscreen-2020-12-23.zip](https://github.com/CR6Community/Marlin/files/5732436/CR-6-touchscreen-2020-12-23.zip)

Can you please retry this with one of these firmware files, depending on the motherboard version you have. If you don't know or got a printer from November 2020 or earlier, you'Il most likely have the Creality v4.5.2 motherboard. 

Assuming we got it fixed I will close this issue now and consider it fixed. If it isn't fixed please provide:
- Logs captured via Pronterface/Octoprint/PuTTY with `M111 S247` entered for debug logging
- Pictures or a video as appropriate

Also, if it is fixed, I still would like to have a confirmation though😀

commit d263457e (2020-11-29): Change M600 filament runout to use the printing pause screen

Should fix #58
closed 2020-12-22 by Sebazzz