CR6Community/Marlin #59: [FR] Home XY in many circumstances

ztynzo, 2020-11-29:
I've noticed that on power failure/resume, filament runout, and pause printing the nozzle seems to stay in place.

In most of these cases it would make sense to home the nozzle to X0 Y0, otherwise the nozzle (often still hot) sits on the print.

shinmai, 2020-11-29:
On a power-loss the position will most likely be lost, and even if it's not, it'll be unreliable since steppers have been off so homing would definitely make sense there; but for the others, Marlin's Nozzle Park feature would work better, I should think.  
And I think it's  actually enabled in the config, not sure why it's not happening (on filament runout at least, haven't really done pauses from the LCD), though. Without checking, my initial suspect is the LCD, but don't quote me on that.

The Mean Well PSU has some beefy caps on it (hence why turning off power and actually powering down has such a lag) I'm now wondering if that could be easily detected and used to move the nozzle up a layer-height or two for PLR like you can with the dedicated "UPS" add-on boards. 🤔 Likely not worth the hassle, but maybe a weekend project for someone wanting to do some Marlin hackin :P But that's getting off topic!

ztynzo, 2020-12-05:
I'll try to do some more testing... I feel like there was one circumstance where once it moved the nozzle off the print, but the other time it didn't.

Unfortunately I'm doing a lot of 24hr prints at the moment, so I don't have a lot of printer down-time to try to replicate issues. :)

Sebazzz, 2020-12-10:
Nozzle park wouldn't work because it works with absolute coordinates. 

> The Mean Well PSU has some beefy caps on it (hence why turning off power and actually powering down has such a lag) I'm now wondering if that could be easily detected and used to move the nozzle up a layer-height or two for PLR like you can with the dedicated "UPS" add-on boards. 🤔

I would imagine that could be done using an Arduino and a inductive sensor. Or perhaps not even an Arduino but one of the expansion ports on the board would even work. The problem is that we don't know the pinout for the Creality boards.

Sebazzz, 2020-12-27:
In CF5:
- Pausing will park the nozzle
- Filament runout will park the nozzle
- After power loss everything is homed then the print continued

For everything else hardware changes are required.

I'Il close this issue now - if there is anything actionable, please reopen this issue or open a new issue.

closed 2020-12-27 by Sebazzz