CR6Community/Marlin #60: [FR] perform PID tuning directly from printer

Allan-N, 2020-11-29:
### Description

Add an option to perform the PID (and PID BED) auto-tuning without using a USB connected device (OctoPrint, Pronterface, etc).

### Feature Workflow

1. Select a PID tuning menu option
2. Provide some feedback as the PID tuning and PID Bed temperature tuning is progresses
3. Save the newly computed values

#### Additional Information

* All printers need some setup.  Why not make things simpler and minimize the need for a connected host.

Sebazzz, 2020-12-10:
Duplicate with #66 - if you have some default PID values we can incorporate, please let us know

hughesjasonk, 2020-12-24:
i have no idea how this is a low priority. It actually is pretty important for people to that go from High temp to low temp and back filaments.

Sebazzz, 2020-12-24:
@hughesjasonk I print PLA and PETG with the same PID without much issues. I imagine that if you need to switch a lot, you have a gcode file ready to swap in the new PID values anyway. 

Feel free to submit a pull request if you feel this is important. Be sure to drop by in our discord because only one person can work at the touch screen part at the same time.

hughesjasonk, 2020-12-24:
> @hughesjasonk I print PLA and PETG with the same PID without much issues. I imagine that if you need to switch a lot, you have a gcode file ready to swap in the new PID values anyway.
>
> Feel free to [...]

I am not looking for an argument. just because it is not important to you doesn't give you the authority to blow off what people need.

as far as the touch screen, if other people and/or myself had the knowledge to update the touch screen it would have already happened.

As far as myself, I am constantly switching between 300ish and 200ish because of the different materials that I use. although that is extreme, Most people going from 200s to 260s is still important to do a PID so you have stable temps while printing.

frenck, 2020-12-25:
@hughesjasonk This is an open-source project. Nobody can get to demand anything from an open-source project. It is free for the grabs, created by people putting in their free spare time in these things, for you to consume for free as well.

If it is that important to you, help out the project and put in your free time. That is how open-source works. I think your comment is out of line and absolutely not called for; it's rather unthankful and impolite even.

Merry Christmas 🎄

hughesjasonk, 2020-12-25:
> @hughesjasonk This is an open-source project. Nobody can get to demand anything from an open-source project. It is free for the grabs, created by people putting in their free spare time in these [...]

Who do you think you are and you probably should stop while you're ahead and butt out of a conversation that you have no part in before I show you what "out of line is"... second, when you "own an open source thread" like this gentleman has, then you absolutely have the responsibility. Second, no one is demanding anything, so yet again, another person in this world, opening their mouth and butting in without understanding the whole situation.

Lastly, some people have the knowledge and most don't. Either help the cause, or sit back down and shut your mouth before I show you what impolite is. Because if you haven't figured it out yet, I'm ready to advocate/fight for the people that can't because they don't have the level of understanding to do.

Oh... merry Christmas cupcake

Sebazzz, 2020-12-25:
@hughesjasonk No need to be so hostile, friend. Please read your response and amend it.

Let me explain my reasoning for why a PID tune is not a high priority opposed to other feature requests:

- Existing workarounds exist, lid using a host or gcode files
- The default PID values work well for the PLA and PETG range
- Users switching often between low and high temp filaments are no rookies and have by now their workflow set up for switching filaments or usually use a host like Octoprint anyway
- If you ofte change you have a gcode file ready anyway to set the correct PID values because a good PID calibration takes quite some time (few minutes at least)

Thank you for understanding.

hughesjasonk, 2020-12-25:
> @hughesjasonk No need to be so hostile, friend. Please read your response and amend it.
>
> Let me explain my reasoning for why a PID tune is not a high priority opposed to other feature requests:
>
> * [...]

And let me explain why I don't agree with you. Also your assumption "Users switching often between low and high temp filaments are no rookies and have by now their workflow set up for switching filaments or usually use a host like Octoprint anyway" is immature and ignorant. There are people that would benefit that do not have a raspberry pi or the room to have a laptop hooked up every time they need to change something.

BLUF: The CR 6 was marketed as a "ready to go platform" so a bunch of inexperienced operators has purchased of which that I have found is not the case. Because of that fact your rational shows laziness and ignorance.

Change a nozzle to a harden nozzle = PID tune
Change a nozzle to brass = PID tune
Change a nozzle to coated copper = PID tune
Print with a temperature sensitive HTPLA that requires you to stay within 2 degrees= PID tune 
printing with temperature sensitive petg = PID tune
printing with temperature sensitive copolymer PETG =PID tune
printing with temperature sensitive PLA+ = PID tune
printing with the worry about layer adhesion = PID

In closing, if you're not going to do it and cop out, step to the side and hand over the group to someone that is willing to help the inexperienced. I personally don't have the experience or the time to reverse engineer what has been done since my IT background is in wireless networking and not programing. otherwise i would have and just did it as it would help a lot of people not able to get to temp in time.

for those reasons is why I feel that it's not a low priority.

MechEng70, 2020-12-25:
@hughesjasonk dude you need to STOP.  This is not a democracy on what this group wants to or not to do.  

If you don't like what this group is doing for free, feel free for fork it off and do what you want.

hughesjasonk, 2020-12-25:
> @hughesjasonk dude you need to STOP. This is not a democracy on what this group wants to or not to do.
> 
> If you don't like what this group is doing for free, feel free for fork it off and do what you want.

There's another one that needs to sit down and shut the F up. If you're not here to help then go back to your corner

hughesjasonk, 2020-12-25:
> @hughesjasonk dude you need to STOP. This is not a democracy on what this group wants to or not to do.
> 
> If you don't like what this group is doing for free, feel free for fork it off and do what you want.

There's another one that needs to sit down and shut the .... up. If you're not here to help then go back to your corner

flofreud, 2020-12-25:
@hughesjasonk and no one indicated that they wouldn't merge a PR for this otherwise the ticket would have been closed. It is just at low priority for now for the maintainers of this project. If someone has an interest in investing their free time into building the feature and open a PR would likely be welcomed. @Sebazzz just pointed out that parallel work on the touch screen is currently really annoying.

Someone wrote a couple of time ago they are doing PID tuning with gcode file on the SD and indicating start and stop of the process with the nozzle LED. @hughesjasonk Probably you could prep files doing this and they can be added to the README or somewhere? Would at least achieve 80% of the proposal and if this gets used a lot, someone will spend the time on making it available through the screen. :)

People need to read the README anyway for the instructions how to install the firmware. Hopefully they would see also the tuning scripts section. 

Currently, development for the interface seems to be more focused on getting basics under control like text placement and not overwriting changes of each other (if I followed the Discord correctly), which indicated to me that this isn't that straight forward.

hughesjasonk, 2020-12-25:
> @hughesjasonk: no one indicated that they wouldn't merge a PR for this otherwise the ticket would have been closed. It is just at low priority for now for the maintainers of this project. If someone has an interest in investing their free time into building the feature and open a PR, it would likely be welcomed. @Sebazzz just pointed out that parallel work on the touch screen is currently really annoying.
> 
> Someone wrote a couple of time ago they are doing PID tuning with gcode file on the SD and indicating start and stop of the process with the nozzle LED. @hughesjasonk Probably you could prep files doing this and they can be added to the README or somewhere? Would at least achieve 80% of the proposal and if this gets used a lot, someone will more likely be motivated to spend the time on making it available through the screen. :)
> 
> People need to read the README anyway for the instructions how to install the firmware. Hopefully they would see also the tuning scripts section.
> 
> Currently, development for the interface seems to be more focused on getting basics under control like text placement and not overwriting changes of each other (if I followed the Discord correctly), which indicated to me that this isn't that straight forward.

Brother! Annoying is not the word for that screen. I have no idea what Creality was thinking. I started reverse engineering of the firmware and my white board got filled up quick. As far as SD card files, that is what I do when I am in a hurry but it still takes some skill and not for the newbie. So I'm not sure how that will help the newbie. Like I said before, this isn't for me. Me advocating is for the newbie that would benefit immensely and I feel the feature is being underestimated. It also should have been included by creality.

nickacker, 2020-12-25:
@hughesjasonk 

This type of behavior will not be tolerated to members of the community who have spent their own free time progressing this project. This FR is marked as low and will stay as low until there is demand for it as we have put out many surveys to get an idea of what the community actually wants as it is a tremendous amount of work(working being done for free) that needs to be completed.

You have been given work arounds until this can be implemented and please remember that this is an open source and free project and as such there is absolutely 0 obligation to work on any FR or even this project if harassment like that continues. I suggest you join the discord if you wish to follow along with development to get an idea of where we're at.

This is not open for discussion. We have accepted this FR and it will remain low for the time being. Just because it is low, doesnt necessarily mean that its going to be last to be completed. This is where its fallen in our gigantic list of other priorities. The comments section is not a place to harass the devs or to even advocate for this FR.

hughesjasonk, 2020-12-25:
> @hughesjasonk
>
> This type of behavior will not be tolerated to members of the community who have spent their own free time progressing this project. This FR is marked as low and will stay as low [...]

Who are you again? If you're going to try and be tough as a keyboard cowboy, stop now and go back to your corner. If you're not here to help and contribute to adding the feature the to screen then sit down and shut up.

nickacker, 2020-12-25:
I am a dev, and I stand by what I said. 

I have reported your account to GH.

Sebazzz, 2020-12-28:
As a stop-gap measure we're now reporting PID tuning results in the 'status messages' bar.

Sebazzz, 2021-01-03:
Current working concept - ignore my pink Comic Sans - that is *not* my specialty. I work on getting the back-end of this screen operational.

Simple: enter temperature, number of cycles, then start it.

[image]

Once done you hear a short buzzer and if successful, the PID values are saved.

Sebazzz, 2021-01-09:
In [#prelease-testing channel](https://discord.com/channels/759603746270609428/791082544899555358/797579692747784232) of our Discord an early build is available if anyone wants to try ([invite link if you haven't joined the Discord earlier](https://discord.gg/RKrxYy3Q9N)).

Sebazzz, 2021-01-26:
Cleaned up user interface by @grobux 

[image]
[image]

With this I now consider this feature implemented. Keep watching the #prerelease-testing channel on Discord for new builds.

**Note:** For bed PID tuning we have issue #138 open.

closed 2021-01-26 by Sebazzz