CR6Community/Marlin #62: [BUG] SD card gets corrupted after a print.

DeivaSR, 2020-12-02:
I am currently running the Community Firmware Release 4 Alpha (v2.0.7.1). I also have the latest touch screen firmware installed.

Using a SanDisk Ultra MicroSD HC 16GB card (see photo). SD card is formatted as FAT32, with Allocation Size of 4096.

These are the steps that I used to reproduce the issue. It is reproducible for me.
1. Insert the SD card into the Windows 10 PC
2. Copy the sliced gcode file which is stored/saved of the PC hard disk to the SD card
3. Eject the SD card before removing the SD card from the PC
4. Insert SD card into CR-6 SE
5. Perform auto-leveling using the touch screen
6. Heat up the bed for PLA (using touch screen)
7. Once heated up, select file to print and begin print.
8. Print completes successfully, and message is displayed on the touch screen with the button "Finish Print". (from this point on, I recorded a video)
9. Tap the "Finish Print" button. Screen goes to printer's main screen.
10. Switch off the power to the printer (using switch).
11. After 10 seconds, switch the printer on
12. As boot boot screen comes up, message pops up:
---------------------------------------
Action Required
Printer stopped unexpectedly. 
Resume previous print?
dwin technology       <<<--- This is not the filename in the first place. But this is not the issue. Please read further.
Resume Cancel (buttons displayed on screen)
---------------------------------------
13. Since print succeeded earlier, tap Cancel (recorded video stops here)
14. Attempting to print the same file again causes the printer to freeze - screen freezes, and print head does not move either. Only way out is to power cycle the printer.
15. Remove the card from the printer and insert it into the Windows PC
16. After the card is read and recognised by the OS,  Windows Explorer pops up.
17. There is a file by the name of PLR in the SD card (this file was not there when it was inserted to the PC in step (1). The file size is 188 bytes (see pic) and the file's timestap is 1/1/2000 (which is surely something that does not seem like is caused by Windows)
18. Attempting to open/read the file using Notepad results in the message  "The file or directory \PLR is corrupt and unreadable. Please run the Chkdsk utility."
19. At this point, the SD card has to be fixed using the Chkdsk utility, or else no files will be able to be written to the SD card.
This issue is reproducible even after formatting the SD card or using a different SD card.
It is not a showstopper. To workaround this, follow the steps below:
1. After each print completes, the printer has to be power cycled and when the Action Required message pops up, tap Cancel
2. Pull out SD card, insert into PC
3. Perform Chkdsk
4. Delete the PLR file
5. And you are good to copy the next gcode file to the SD card for the next print.
Anyone else seen this problem? And should I files this in github?
Video and screenshots attached.

[image]

[image]

[image]

Video: https://youtu.be/b2nk_uuOr68

Sebazzz, 2020-12-02:
Can you try this build and check if it improves things: https://github.com/CR6Community/Marlin/files/5619260/firmware-20201201-000916.zip

DeivaSR, 2020-12-03:
> Can you try this build and check if it improves things: https://github.com/CR6Community/Marlin/files/5619260/firmware-20201201-000916.zip

I tried this build today. Seems like the SD card corruption is still taking place. Exactly the same as what was reported in the initial bug report.

Sebazzz, 2020-12-03:
Sorry, I can't reproduce it. Still assuming for now it doesn't happen on stock hardware. 
For now I suggest disabling power loss recovery.

@CR6Community/developers Any idea?

JonBradbury, 2020-12-03:
Hi Sebastiaan 

Regarding power loss recovery.

At a guess, I’d say card corruption occurs when a power outage happens during a write operation. IIRC it uses the SD card to record the current position of the print because writing it to EEPROM is too slow, causing blobs in vase mode. If it powered down while doing an SD card write, well, corruption is possible. But we should check the implementation of the write operation first. I would expect it to open the file, write the data, then close the file at every update; this is going to give as small as possible a window for a power failure to cause corruption. Knowing Creality they probably open the file at the start of the print, update continuously, then close at the end of the print.

If I’m right, it means your failure to reproduce it is due to not powering down the printer as the SD card write is in progress. That’d be tricky to achieve.

Cheers
JonB

Sebazzz, 2020-12-03:
@JonBradbury That would make sense if this is Creality-based, but in this case pretty much everything has been written by us or replaced by a default Marlin implementation and in this case power loss recovery is default Marlin. Perhaps @InsanityAutomation has any idea, he is a bit closer to the fire.

DeivaSR, 2020-12-04:
@Sebazzz @JonBradbury Could it be possible that, in some cases (e.g. in my case), after printing completes successfully, the file being written to the SD card is not closed (thus the PLR file remaining in the SD Card), resulting in the corruption?

I found that when Windows prompts that the SD card has some errors, and I perform a Scan and Fix, the PLR file gets removed by Windows during the fix process. Once I have fixed the SD Card and reinsert it to the printer, performing a power cycle will go through without the "Action required... Printer stopped unexpectedly message"

I will add more info to the bug report on what I mentioned above.

InsanityAutomation, 2020-12-04:
PLR by nature is abusive to sd cards. The function with a continually open write handle defeats the internal wear leveling of the card. It causes shakey cards to fail very quickly and easily. Limiting writes with a z-change limit is critical in mitigation.

Sebazzz, 2020-12-04:
> Could it be possible that, in some cases (e.g. in my case), after printing completes successfully, the file handle to the PLR file that is being written to the SD card is not closed (thus the PLR file remaining in the SD Card), resulting in the corruption?

Well, the fact is that the CR-6 touch screen firmware has no option to "release media". But... I have never used that either on other printers and yanked that card out of it without consideration (and without issues) 😀

@DeivaSR: Can you test this:

1. Print like in your issue described
2. *Before* removing the card from your printer, issue gcode M22 from a serial terminal.

Sebazzz, 2020-12-10:
Closed due to inactivity.

closed 2020-12-10 by Sebazzz