CR6Community/Marlin #64: [BUG] Serial port for pronterface or octoprint not working in branch extui-btt-skr-stock-tft

NickDevoctomy, 2020-12-07:
### Bug Description

After flashing extui-btt-skr-stock-tft branch to a BTT SKR CR6, the serial port is non functional and showing as an unknown USB device in Windows.  After some looking about it seems it may be down to the way the serial ports are configured in Marlin.

### Configuration Files

Config files in extui-btt-skr-stock-tft branch

### Steps to Reproduce

1. Flash extui-btt-skr-stock-tft
2. Connect USB to Windows device
3. Check device manager to see if the device has been detected as a virtual serial port or unknown device

**Expected behavior:**

Should expect a virtual serial device to be detected

**Actual behavior:**

Uknown usb device is detected

Sebazzz, 2020-12-08:
You need to install the drivers. This is not a firmware bug - in fact - the firmware has no control over how the USB-to-serial chip identifies itself. Other people have successfully flashed this branch - so that also leads me to think it is not a firmware issue.

closed 2020-12-08 by Sebazzz

NickDevoctomy, 2020-12-08:
Okay, I'm not sure I'm aware of any drivers? Also no other printer board that I've connected to has required drivers.

I was lead to believe it is the firmware by BTT themselves who have suggested the SERIAL_PORT definitions being incorrect when the problem has been experienced with other boards of theirs.

The CR6 is different in this case as the LCD screen is using the second serial port.

Have you managed to connect to it via usb using this firmware?

Sebazzz, 2020-12-08:
Yes, without issue.

NickDevoctomy, 2020-12-08:
In Windows, and you installed some drivers?

Sebazzz, 2020-12-08:
Yes, in Windows. No drivers, but I suppose they were installed automatically because I used a similar board (in the BIQU B1) earlier.

NickDevoctomy, 2020-12-08:
Okay, well everything I'm finding online about this suggests it's a Marlin issue and the SERIAL_PORT definitions as these literally control how the board communicates with the host. So if it works for you im now under the impression that I have a faulty board as there are no drivers advertised for either boards that I can see at a glance. This sort of thing is usual part of Windows.

Sebazzz, 2020-12-08:
Can you make a screenshot of device manager? Do you have the issue on stock firmware too?

NickDevoctomy, 2020-12-08:
Sure thing, I'll get some screenshots this morning. I hadn't tried connecting to Pronterface as it was unusable, crashed when attempting to print from the SD card for example.

NickDevoctomy, 2020-12-08:
Okay great, not sure what happened but that actually fixed it. Done quick sanity check as it was, USB serial from laptop didn't work (didn't take a screenshot though), flashed stock firmware, USB worked, flashed extui branch back and now it works with that too. Awesome.

NickDevoctomy, 2020-12-08:
Okay so I was wrong about it working perfectly now, I seem to have some other kind of issue which I can only assume is hardware related and may have been the cause of this original problem.  This probably has nothing to do with the firmware but am writing it here just incase.

Basically when I plug in a USB lead to my laptop, after about 15 seconds or so, an audible alarm sound starts.  Now I have no idea if it's an actual alarm, or a fan making a weird noise, but it is entirely related to having a USB lead connected to my laptop.

I thought it may have had something to do with me disabling the filament runout sensor, as that's the only difference between my image and the btt extui branch. So I re-enabled it again and tested it, but the "alarm" continues to sound.  If the machine is printing when I plug in the USB, the alarm starts and the printer crashes, so it's not just a sound, it's causing some weird issues.

I should confirm that is with or without the power line being present on the USB or not.  Unplugging the lead does not stop the sound until I power down the machine.

Also... If I connect it when the printer is cool, (20c ~), it doesn't sound the "alarm".  

A video of the issue is here,

https://www.dropbox.com/s/0iyfd4qqpuyv352/20201208_123418.mp4?dl=0

tldr;

1.  "Alarm" sounds when the following occurs,
  a. printer plugged into laptop via usb (with or without power line present)
  b. printer hot end being hot (but not actively heating, and not directly linked to the fan)
  c. after about 10-15 seconds of being plugged in

Sebazzz, 2020-12-08:
Very odd. There is nothing that can give an audible noise from the hot-end except the fan - that alarm is the fan.

Beeble2695, 2020-12-08:
Are you using an USB cable with the 5V pin being taped, or an unmodified one?

NickDevoctomy, 2020-12-08:
The part cooling fan does have a tendency to rattle, so I could replace it, but that doesn't explain the crashing.

NickDevoctomy, 2020-12-08:
And yup, I tried with and without the 5v removed. Sorry can't see all messages on my mobile.

Beeble2695, 2020-12-08:
Yeah, that was me not reading ;)

I assume the printer is connected to a grounded wall socket?

NickDevoctomy, 2020-12-08:
It should be, I mean I'm in my house.  But I'll try a different socket in a bit to see if that makes any difference.  I'll report back with my findings...

NickDevoctomy, 2020-12-08:
Okay just tried a few other sockets, no change.

Also tried with the USB lead plugged into the printer, but not plugged into anything else, and that didn't set it off (as you'd expect).

64Spaces, 2020-12-08:
Just a sidenote. I’ve noticed that PWM fans on my Raspberry Pi’s will make noises like this, when running at less than 100%.

NickDevoctomy, 2020-12-08:
Interesting... Just tried connected to my Raspberry Pi 4, no alarm sound.  Whilst running I connected to my Surfacebook, no alarm sound.  Turn printer off, and on again whilst connected to Surfacebook, alarm starts.

Basically it only seems to make the noise when connected to my laptop, so that means my laptop has a power issue?

Beeble2695, 2020-12-08:
Could you try another USB port? see if that matters?

NickDevoctomy, 2020-12-08:
Yeah only have 2 on the laptop, tried both and they all set it off.

Tried all 4 on the pi and none of them back the noise.

Sebazzz, 2020-12-08:
I must say that I have observed this also a few times, that connecting or disconnecting can result in a reboot. It doesn't always happen. It isn't the serial handshake causing it - it is the physical connection. So I've just stopped connecting or disconnecting when I have something important running. Except for this - it is completely stable for me.

NickDevoctomy, 2020-12-08:
Just tried another laptop, and it didn't make the noise. COM port appeared as expected in device manager too.

Beeble2695, 2020-12-08:
Just like those cases with people who've had those metal covered laptops and felt a tingling sensation when touching it. Was mostly due to it being a non-grounded AC connection. So the person would act as a way to ground.

NickDevoctomy, 2020-12-08:
Aaah! I get that sensation when running my fingers on the surface to the side of the trackpad, almost feels like some weird static friction.

So my laptop isn't grounded properly?

Beeble2695, 2020-12-08:
Most likely yes, and thats the fault of the manufacturer.
Dell even has an article on it
https://www.delltechnologies.com/en-us/blog/notebook-tingle-sensation-what-39-s-going-on/

Beeble2695, 2020-12-08:
I wouldn't be suprised if that sensation is gone when you do connect it to the printer

NickDevoctomy, 2020-12-08:
Well that's been an eye opener, thanks for all the help everyone, I'll just have to use a different laptop for connecting it.

damiani8, 2020-12-08:
My fan makes that exact noise on mine with pronterface running.