CR6Community/Marlin #71: [BUG] Homing failed

giorgiomanieri, 2020-12-10:
### Bug Description

Home failed message during the homing after ABL

### Configuration Files

**Required:** Community firmware alpha 4

### Steps to Reproduce

1. Complete auto bed leveling
2. Start a print from Octoprint

**Expected behavior:**

**Actual behavior:**

After the bed leveling, the printer starts the homing and after that show Home failed message. 
If I shutdown and turn on again before the starting print, the problem doesn't persists. 
#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

Sebazzz, 2020-12-10:
Without logs I can't do anything.

InsanityAutomation, 2020-12-13:
I have a code base here with the strain gauge and tare code refactored. Can you give my changes a try and see if its any more stable?

https://github.com/InsanityAutomation/Marlin/tree/CR-6Devel

leyton01, 2020-12-21:
I am having this issue as well. Would a log file help?

Sebazzz, 2020-12-21:
Can you try the changes from @InsanityAutomation ?

MericasMeuth, 2020-12-21:
> Can you try the changes from @InsanityAutomation ?

Where do I find these changes?

InsanityAutomation, 2020-12-21:
> 
> 
> > Can you try the changes from @InsanityAutomation ?
> 
> Where do I find these changes?

https://github.com/InsanityAutomation/Marlin/tree/CR-6Devel

Sebazzz, 2020-12-22:
[Creality v4.5.2 motherboard firmware-crealityv452-20201223-002158.zip](https://github.com/CR6Community/Marlin/files/5732408/firmware-crealityv452-20201223-002158.zip)
[Creality v4.5.3 motherboard firmware-crealityv453-20201223-002246.zip](https://github.com/CR6Community/Marlin/files/5732406/firmware-crealityv453-20201223-002246.zip)
[Touch screen firmware CR-6-touchscreen-2020-12-23.zip](https://github.com/CR6Community/Marlin/files/5732436/CR-6-touchscreen-2020-12-23.zip)

Can you please retry this with one of these firmware files, depending on the motherboard version you have. If you don't know or got a printer from November 2020 or earlier, you'Il most likely have the Creality v4.5.2 motherboard. 

Assuming we got it fixed I will close this issue now and consider it fixed. If it isn't fixed please provide:
- Logs captured via Pronterface/Octoprint/PuTTY with `M111 S247` entered for debug logging
- Pictures or a video as appropriate

Also, if it is fixed, I still would like to have a confirmation though😀

closed 2020-12-22 by Sebazzz

giorgiomanieri, 2020-12-23:
Solved for me

leyton01, 2020-12-23:
Fixed for me in v5 pre-release