CR6Community/Marlin #73: [FR] Please consider implementing Marlin's REDUCED_ACCURACY_WARNING on this display

Thinkersbluff, 2020-12-12:
When my Ender 3 needs to be homed, Marlin 4.0.7.2 alternately displays "??" and "0.0" or "??", Blank, "0.0", (whatever it takes to make it look as if the displayed values are flashing)

Since Marlin will happily drive an axis into the stops and keep going, when Marlin "thinks" the axis is currently at a position which it is not, it would be a kindness to remind the user when Marlin really has no idea where the axes presently are. 
e.g.:
- From when the machine is first powered up, until it is first homed; 
- Whenever "Disable Steppers" is pressed; 
- Whenever the non-zero DEFAULT_STEPPER_DEACTIVE_TIME has been reached (at least for those axes set with disable "TRUE".

I see that Configuration.h actually has a special setting to disable that flashing warning, so it may be a consideration to design this feature in a way that could also be overridden with ; this setting, though I could live without it.
"// Turn off the display blinking that warns about possible accuracy reduction
//#define DISABLE_REDUCED_ACCURACY_WARNING"

Sebazzz, 2020-12-13:
It is not very easy to do due to the way the DWIN touch screen is configured with the numeric inputs. Also chiming in @grobux here.

I don't think the numerical inputs have any native way to display an unknown position. However, using stack pointers we might be able to change the color.

InsanityAutomation, 2020-12-13:
Only thing I can think of to do easily is flash 999 / value based on the same blink code from the ultralcd 920 code

Thinkersbluff, 2020-12-13:
If we are not able to mimic the LCD display's behaviour (i.e. flashing "??") are there other techniques used to caution users of TFT displays, when the displayed positional data may not be correct?

If you can change colour, for instance, changing the displayed value to the same colour as the background would serve to blank their display.  Toggling colours could serve to make the values seem to flash.  Is the difficulty a matter of knowing when to give this feedback or just how to present a particular type of cautionary message?

Sebazzz, 2020-12-24:
Implemented now. Unknown axis are shown as red, and when homed back to white.

Sebazzz, 2020-12-26:
Closing because this is implemented and ready for release in CF5 beta.

closed 2020-12-26 by Sebazzz