CR6Community/Marlin #74: [BUG] No print resume after pause or stop at height

ValChe82, 2020-12-14:
When a print is paused or stop at height in cura (eg: to change filament) the print do not start after clic on resume. Even when sending "M24-Start or Resume SD print" on Pronterface.

There is the same behavior if power is lost.

Sebazzz, 2020-12-22:
[Creality v4.5.2 motherboard firmware-crealityv452-20201223-002158.zip](https://github.com/CR6Community/Marlin/files/5732408/firmware-crealityv452-20201223-002158.zip)
[Creality v4.5.3 motherboard firmware-crealityv453-20201223-002246.zip](https://github.com/CR6Community/Marlin/files/5732406/firmware-crealityv453-20201223-002246.zip)
[Touch screen firmware CR-6-touchscreen-2020-12-23.zip](https://github.com/CR6Community/Marlin/files/5732436/CR-6-touchscreen-2020-12-23.zip)

Can you please retry this with one of these firmware files, depending on the motherboard version you have. If you don't know or got a printer from November 2020 or earlier, you'Il most likely have the Creality v4.5.2 motherboard. 

Assuming we got it fixed I will close this issue now and consider it fixed. If it isn't fixed please provide:
- Logs captured via Pronterface/Octoprint/PuTTY with `M111 S247` entered for debug logging
- Pictures or a video as appropriate

Also, if it is fixed, I still would like to have a confirmation though😀

closed 2020-12-22 by Sebazzz

ValChe82, 2021-01-04:
Hello,

Thank you for the feedback and the new firmware. Yes my mainboard is a
4.5.2. I actually directly switched to the Special New Year edition.

So far so good! I tested as many interruption scenarios as possible which
can happen, pause, stop, power shortage, etc..

I haven't found any issue at all. Very good job from de Dev Community team.

Meilleures salutations

Valentin Chevre