CR6Community/Marlin #77: Can't hold extruder temperature

grimlinger90, 2020-12-17:
Hi, since i upgraded to the community firmware, setting the temperature on the extruder is a bit weird.
I can pre heat the extruder and bed, and then I load a print job from SD, extruder temp will be back at 0°C, bed stays at 60°C . When the job started I can change the temperature to what I want and it holds it. 

Installed octoprint couple days back , same thing, and I cannot control the temperature of my extruder at all with octoprint, no matter what I do it will go back to 0°C . 

Using community firmware release 4 alpha on creality cr6-se , Cura 4.8 on win10, with a raspberry b+ with octoprint 0.17.0. Thanks !

Sebazzz, 2020-12-17:
Do you use `G29` in your start gcode? Replace it with `M420 S1`.

grimlinger90, 2020-12-17:
Ok will try this thank you ! Failed to mention that when it's on a job, I can adjust the temp value in octoprint and the value sticks. 
Next print job ill change the G29 with M420 S1 

"G28 ;Home

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position"

Unostot, 2020-12-17:
Did you update the touchscreen as well?
I'm not sure how it is working together, but perhaps the touchscreen is resetting every few seconds.
In the alpha its statet that the touchscreen firmware needs to be updated too...

I'm using octoprint, firmware 4alpha and the updated touchscreen with no problems over a week now. Despite some problems with "double leveling" (see other issue) its working good.

A PID Autotune did help to hold the temperature more exact (less jitter, but was good before, too), but I did not have problems with the temperature resetting...

Greetings,
Uno

Sebazzz, 2020-12-22:
[Creality v4.5.2 motherboard firmware-crealityv452-20201223-002158.zip](https://github.com/CR6Community/Marlin/files/5732408/firmware-crealityv452-20201223-002158.zip)
[Creality v4.5.3 motherboard firmware-crealityv453-20201223-002246.zip](https://github.com/CR6Community/Marlin/files/5732406/firmware-crealityv453-20201223-002246.zip)
[Touch screen firmware CR-6-touchscreen-2020-12-23.zip](https://github.com/CR6Community/Marlin/files/5732436/CR-6-touchscreen-2020-12-23.zip)
Can you please retry this with one of these firmware files, depending on the motherboard version you have. If you don't know or got a printer from November 2020 or earlier, you'Il most likely have the Creality v4.5.2 motherboard. 

Assuming we got it fixed I will close this issue now and consider it fixed. If it isn't fixed please provide:
- Logs captured via Pronterface/Octoprint/PuTTY with `M111 S247` entered for debug logging
- Pictures or a video as appropriate

Also, if it is fixed, I still would like to have a confirmation though😀

closed 2020-12-22 by Sebazzz

Grimmvg, 2020-12-22:
awesome ! hope to find some time this week still and try it out :) this is great thanks !