CR6Community/Marlin #78: Creality CR6-SE Cannot Scroll on SD Screen

Thebuz, 2020-12-21:
Creality CR6-SE
Cannot Scroll on SD Screen

I can view and print anything on the 1st page and open folders. but pressing the arrows to scroll through pages is not working

Sebazzz, 2020-12-21:
Insufficient information, I don't know how you expect me to say anything about this without more information.

Thebuz, 2020-12-21:
I don't know what more there is to say.

Put an SD card into the CR6-SE with 20 gcode files on it

Press print on the CR6-SE touch screen 

It displays the 1st Files that are on the SD card

Press the arrow on the touch sceeen for the next page, nothing happens.

Sebazzz, 2020-12-21:
I'm sorry for being blunt, but the information you're giving is really scarce. Things like the firmware bin files you've flashed for the touch screen and printer itself, logging using M111 etc etc.

Thebuz, 2020-12-21:
Ah ok.

CR-6-touchscreen-2020-11-06

firmware-20201109-211912-hotfix3

I can't use logging as I don't want to risk damaging my Pi 4 with the power issue untill i get a modified USB cable, which is why I am printing from the SD Card.

Sebazzz, 2020-12-21:
I cannot reproduce the issue. I suggest you reflash both firmwares.

Can you describe the issue further? Can you only one file inside a folder, or only one folder or file at all? Can you include some pictures?

@Nushio is running pretty much CF4 alpha screen firmware - can you confirm @Nushio?

Nushio, 2020-12-22:
My printer's been printing non-stop (Xmas gifts). I have a "window" in about 4 hours when I can load it with files and try to replicate before I start my next 18 hour print :-P

Thebuz, 2020-12-22:
I am not sure what the issue was, i could reflash the touchscreen software, but with Marline i had to downgrade to an older version then reflashed, but that fixed all the SD card issues. thank you for your time

closed 2020-12-22 by Thebuz