CR6Community/Marlin #80: Ability for the screen to go on sleep

SpiCraftsman, 2020-12-21:
Hello, 

The official firmware has the ability for the screen to turn off after x seconds(which is bad because you can't change it anywhere) it you be useful having something like that.

Thanks

Sebazzz, 2020-12-21:
I need to look further into the documentation of DGUS whether it is possible to set this dynamically. I'm afraid not though.

Sebazzz, 2020-12-22:
[image]

It should be possible by writing to register 0x82.

Sebazzz, 2020-12-24:
This is now implemented in the touch screen, temporary locations are here:

[image]

@grobux is further refining the UX for this

Screen brightness is reduced to 10% on stand-by, and a mute function has been implemented.

grobux, 2020-12-25:
Screen updated, here's a preview:
[image]

Sebazzz, 2020-12-26:
I'm closing this now because it has been implemented and is ready for CF5 beta. In addition, I've implemented a control for setting the screen stand-by brightness value. @grobux is doing the final UI polishing.

closed 2020-12-26 by Sebazzz