CR6Community/Marlin #83: [FR] Determine a final 'good' default settings for movement kinematics

Sebazzz, 2020-12-23:
Determine a good default set for movement kinematics. 

Think of things like:
- Linear advance enabled/disabled
- S curve acceleration
- Classic jerk vs junction deviation (and which specific jerk and junction deviation values)
- Acceleration

The current values are fairly conservative. We might be able to squeeze something more out of this printer.

frenck, 2020-12-23:
Acceleration tuning using: 

<https://teachingtechyt.github.io/calibration.html>

[image]

[image]

Printed using worthless cheap Creality PLA, but nevertheless, there aren't really notable differences in terms of acceleration (maybe on 800 it becomes a little more sloppy).

I would say anything between 400-700 will do. The current default acceleration of 500 seems fine.

Sebazzz, 2020-12-23:
@frenck Thanks

Related issue to take into account: 17342

frenck, 2020-12-27:
Finished some larger prints now, using S-Curve Acceleration, which works pretty well for me. Pushed the printer to 110mm/sec with these tests and still got a really really nice print.

Please note, in order to enable `S_CURVE_ACCELERATION`, `EXPERIMENTAL_SCURVE` must be enabled in the `LIN_ADVANCE` section as well, as using S-Curve with Linear Advance is currently experimental.

Furthermore, I've lowered the default retraction acceleration. Currently, it is set to 1000mm/sec, which I have lowered to 750 right now, might even lower it to 500. I haven't noticed much difference in print quality but it just seems high and I feel like the extruder is more "clam" now in terms of sound it produces.

Sebazzz, 2020-12-27:
S-curve is more about vibrations isn't it?

frenck, 2020-12-27:
Well yes/no... its motion planning, causing a more smooth motion (instead of a constant acceleration value).

So, yeah, less vibration is a result, the machine sounds more "relaxed", and the longer sides look smoother (as a first impression). That smoothness is probably caused by the less "instant" acceleration.

As a downside... Dunno? I gues it is a tad slower? And of course experimental at this point.

frenck, 2020-12-27:
Found some more information about this subject here: <https://github.com/synthetos/TinyG/wiki/Jerk-Controlled-Motion-Explained>

chriskoups, 2021-01-02:
Hello!
I am interested also to enable s_curve as in my CR-10 made a huge difference!
Something that i noticed and reported on the screen repo was that the Default E jerk for RC5 was 0 (command M205 reported E0.0) can you confirm what you have @frenck ?
If is at 0, then you can improve a lot the quality by setting it to 5, as it defaults from quality.
Jerk helps with the transition of acceleration and improve extrusion.

Sebazzz, 2021-01-02:
Default E-jerk is indeed set to 5.0.

chriskoups, 2021-01-02:
Yes, saw that! I set the linear advance to 0.5 and seems to produce perfect results. I am using SuperSlicer.

Thinkersbluff, 2021-02-03:
This upstream issue looks interesting & related: https://github.com/MarlinFirmware/Marlin/issues/19217  

Might be more of an issue for the MAX folks?

chriskoups, 2021-02-13:
Hello Folks,

So this weekend I aim to find the printers limits just to see how much we can safely push it.
Currently I have those settings but with a 0.25mm height limit it needs at least 230C on the nozzle otherwise on flat big surfaces the extruder cannot keep up.
[image]
[image]

I found out that the maximum Y speed is practically around 200, more than that I have vibrations, The X speed can go up to 300 but in practise it almost never go up there.

My biggest problem is with the accelerations... I increased them so much but it seems that the estimated time in SuperSlicer is not accurate and is at least 40% less... I made sure that my EEPROM settings are the same as the slicer... 
Is it possible that the power is not enough to the motors? But that must introduce skipping.....
I adjusted the VREF to 1.1V for X-Y, 1.2V for E-Z, They were around 0.1-0.15 more.

Sebazzz, 2021-02-13:
@chriskoups Thank you for your work on pushing this printer. Have you enabled in your slicer that the firmware accelerations need to be overriden and not just used for time estimations?

chriskoups, 2021-02-13:
@Sebazzz No As I entered the values in the EEPROM.
I will also try now "also emit limits to G-Code" to see the difference if any.

chriskoups, 2021-02-13:
Nothing changed.... I only found that maximum extruding acceleration of 3000 can introduce skipping if they are multiple small parts, reduced to 1500.
I found that line in configuration that may introduce delay:
#define SERVO_DELAY { 300 }
But seems to small to be the problem

chriskoups, 2021-02-14:
After reading a bit the configure file, I found that the problem is with Ejerk as I use Linear advance. I would send my maximum safe settings in the afternoon after running few tests. I had it at 5 but introduced delays in prints. I pushed it at 25 and it responded nicely but I want to see if a further reduction keeps the same results.

Something that I have to mention, and is really important as 99% of users do it wrong, is the E maximum feedrate as it is set to 25mm/s which is logical as the extrusion volume cannot be more than 12ish mm^3/s (equivalent to 5mm/s) but any retraction with speed more than 25mm/s will be capped at 25mm/s. If we increased the allowed speed in EEPROM, then we must cap it on the slicer based on the extrusion volume.

Sebazzz, 2021-02-14:
Sounds good, looking forward to your results.

chriskoups, 2021-02-14:
Results from this weekend. They seems stable for small parts and a big rectangular item that I printed.
I have not tested lithoplanes, vase mode or character printings.
[image]
The maximum Feedrates are around 10% less from when I hear the axes making noise.
The acceleration and jerk are around 20% from when the axes start losing steps.
The E-jerk is a particular setting as i could see that is used only when linear advance is enabled. I have mine set to 0.65.

The maximum reliable speeds that I was able to achieve are:
[image]
With a layer of 0.15, a temperature around 210 is ok but for 0.25 I had to increase to 230. The flow of my stock extruder cannot go more than 12mm^3/s with 230 degrees.

Please test and report. I think will be good to have the optimal and maximum settings.
I think for optimal/safe settings I will reduce acceleration and speeds by 30%.

Sebazzz, 2021-02-14:
Which firmware changes did you do or were you able to do everything from the slicer and gcode?

chriskoups, 2021-02-14:
I am using this firmware as you build it:
CF5Beta-cr6-se-v4.5.2-mb-2020-12-30-14-19.zip

I introduce all the acceleration, feedrates and jerk through Gcode to the EEPROM.
I assume the Slicer can set those per print for testing if the testers do not want to put them in the EEPROM.

Something that I would suggest (maybe it changed in the latest build) is the bedlevelling and homing temperatures. I would suggest to pump the temp from 150 to 200. The material can be hard outside the nozzle and the homing may level few mm over the bed. Happened few time today over the tests.

Sebazzz, 2021-02-16:
I've made a separate branch for these kinematics changes. After flashing the firmware and resetting to factory settings the first thing that stands out after leveling, is that the stepper noise has increased. This also happens during certain movements between homing and starting the actual skirt. 

Not that I mind it, but it is is to note that it might "scare" some users away. Printing a benchy now.

chriskoups, 2021-02-16:
Is the branch public? I could not find it.
Have you noticed any axis in particular that was making noise? 
I hope that X will be quite but Y and Z can be noise if they are not oiled or if the tension is too much.

While printing, what was the noise level? The vibrations?

As I said in my previous comment, those are the maximum that I feel comfortable to use. For the default, I would suggest:
X acceleration 2000
Y acceleration 2000
Z acceleration 300
E acceleration 3000
Extruding acceleration 1000
Retracting acceleration 1500
Travelling acceleration 1500

The feedrate and jerk must be ok....

Looking forward to your feedback.

Sebazzz, 2021-02-16:
I didn't push yet 🙂 I've just pushed it.

chriskoups, 2021-02-16:
em.... Extruding, Retracting and Travelling are set 3000, 5000 and 1500 respectively.... On my printer that was too much... I do not know if you mistype them 😱

Sebazzz, 2021-02-16:
I guess I misinterpreted that from the screenshots 😅 Still, benchy seems to be going pretty well so far.

chriskoups, 2021-02-17:
Any exciting results? A benchy?

Sebazzz, 2021-02-17:
Printed in silk PLA, so be aware that any imperfections show up ten times. 

[image]
[image]

Looks good. Not better, not worse. Gcode file used (settings are included):

[PLA Fast_(1h38m).zip](https://github.com/CR6Community/Marlin/files/5996817/PLA.Fast_.1h38m.zip)

chriskoups, 2021-02-17:
Looking in the gcode, I think you override some of the changes with the slicer..... 🤔 
M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk

Sebazzz, 2021-02-17:
I'm not printing benchies late at night anymore 😅 Retrying.

Sebazzz, 2021-02-17:
[image]
[image]
[image]`

Very similar, if not the same.

Printer sounds way more aggressive though, I imagine some users might be intimidated by these defaults. 🙂

chriskoups, 2021-02-17:
To be expected, you were printing to the limits...
What was the actual printing time of both? And their printing speeds?
Have you try to go at 100-125m/s?

Thinkersbluff, 2021-03-10:
I like to uncomment 
//#define ADAPTIVE_STEP_SMOOTHING in Configuration_adv.h 

I see it is commented out by default in the CF configs.  
Do you know of a problem with it, or could that be something else to consider?

Sebazzz, 2021-03-10:
I see some issues if I search the Marlin discord that `ADAPTIVE_STEP_SMOOTHING` might cause some probe accuracy issues, which is definitely something way can't use on this printer. Unclear if it is BLTouch only. It is a fairly new feature it appears?

Ranney1, 2021-03-10:
I've read the same. Example: https://github.com/MarlinFirmware/Marlin/issues/18598

Maybe trying to do some probe tests with and without this function enabled?

Thinkersbluff, 2021-03-10:
Wow.  I had no idea.  Thanks for the heads up.  Looks like CPU loading is an issue, also.  Maybe not as helpful a feature as the comments in the config file make it out to be.  I will try without it for a while and see whether the "singing" returns as I draw round things...